zephyrproject-rtos/zephyr #26170: HEXDUMP log gives warning when logging function name.

Emil-Gydesen-Bose, 2020-06-13:
**Describe the bug**
At least for `BT_HEXDUMP_DBG` the log statement does not include the function name, and if you add the function name as part of the string supplied by the caller, then this warning: `WARNING:EMBEDDED_FUNCTION_NAME: Prefer using '"%s...", __func__' to using 'my_func', this function's name, in a string` appears. Since it's not possible add formatting to HEX_DUMP log strings (without additional steps), this warning cannot easily be dealt with. 

**To Reproduce**
Create a function that logs it's own name using e.g. `BT_HEXDUMP_DBG`:

```
void my_func(void *data, uint16_t data_len)
{
    BT_HEXDUMP_DBG(data, data_len, "my_func was called");
}
```
Log should be something like

```
[00:01:03.770,000] <dbg> bt_my_file.my_func was called
...
```

**Expected behavior**
That either there is no warning, or that the log statement automatically includes the function name (similar to `BT_DBG`). 

**Impact**
Very low as it is just a warning. 

**Additional context**
Add any other context about the problem here.

nordic-krch, 2020-06-15:
Unfortunately, it is hexdump limitation that it cannot have function prefix. Added clarification to kconfig and doc.

Emil-Gydesen-Bose, 2020-06-15:
> Unfortunately, it is hexdump limitation that it cannot have function prefix. Added clarification to kconfig and doc.

The limitation is fine, but then it should perhaps be "legal" for the application to log the function name (even as a literal string rather than `__func__`). Not sure if that can easily (or at all) be done for the check_compliance script.

Emil-Gydesen-Bose, 2020-06-16:
Solutions for this issue has been discussed in #26178

carlescufi, 2020-06-16:
Reopening as enhancement as per [this comment](https://github.com/zephyrproject-rtos/zephyr/pull/26178#issuecomment-644741673)

PR #26178 (2020-06-16): logging: Clarify that hexdump messages have no function prefix
It was not clear that hexdump messages does not support prepending
with function name. Added clarification.

Fixes #26170.
closed 2021-04-19 by nashif