zephyrproject-rtos/zephyr #26176: Add bt_is_id_valid functionality

jeremyherbert, 2020-06-15:
**Is your enhancement proposal related to a problem? Please describe.**
When a device that uses multiple identities is programmed and powered on for the first time, it may need to check how many IDs are currently valid so that it can initialise any missing ones. I can't seem to find a good way to do this at the moment.

**Describe the solution you'd like**
Any of the following functions would be suitable (in `hci_core.c`):

1. Add a new function
```
bool bt_is_id_valid(u8_t id) {
  return id < bt_dev.id_count;
}
```

2. Change `bt_id_get` so that the `*addrs` argument can be NULL
```
void bt_id_get(bt_addr_le_t *addrs, size_t *count)
{
	size_t to_copy = MIN(*count, bt_dev.id_count);

	if (addrs != NULL) {
		memcpy(addrs, bt_dev.id_addr, to_copy * sizeof(bt_addr_le_t));
	}
	*count = to_copy;
}
```

Actually, both of these changes could be useful if added. I am happy to create a PR myself if that would be useful.

carlescufi, 2020-06-15:
@jeremyherbert please go ahead and create a PR and then we can discuss it there directly.

jhedberg, 2024-12-16:
This seems to have been already implemented

closed 2024-12-16 by jhedberg