zephyrproject-rtos/zephyr #26179: devicetree: Missing support of unquoted strings

DeLat, 2020-06-15:
**Is your enhancement proposal related to a problem? Please describe.**
To avoid runtime configuration of driver, it's useful to place some data to configuration output section of driver. And because hardware can have multiple instances of driver with multiple configuration sets, it's necessary to generate named sections in linker script by concatenation with label of device tree node or other device tree string property. For example I want to reference BOS descriptor in rewritten USB stack from device tree overlay:

```
&usb1 {
    primary_set: usbset0 {
        label = "primary";
        compatible = "usb-set";
        vid = <0x2FE3>;
        pid = <0x0100>;
        manufacturer = "MANUFACTURER";
        product = "PRODUCT";
        bos = "usb_config_bos_primary";
        max-power = <50>;
    }
}
```

With following pre-processor code in linker script to generate list of input sections selectors for each of USB controller instance:

```
#define USB_BOS_CAP_SECTION(inst, id) \
                KEEP(*(STRINGIFY(.usb.bos.id)))

#define USB_BOS_CAP(i, inst) \
                USB_BOS_CAP_SECTION(inst, DT_PROP_BY_IDX(inst, bos, i))

…
UTIL_LISTIFY(DT_PROP_LEN(inst, bos), USB_BOS_CAP, inst)
…
```

This allow to place BOS structure where we need it during compilation. BOS structure can be declared in _.usb.bos.usb_config_bos_primary_ input section

Second problem that can be solved with unquoted strings in device tree is using this strings to generate names of variables, this allows to avoid runtime configuration that can be made during compilation. For example, device configuration structure can use it to reference some structures in user application without placing it to specific driver output section. Like OS descriptors in USB stack again.

**Describe the solution you'd like**
To avoid conflicts with #21369 I propose to add special types for binding YAML, like "token" and "token-array" to generate unquoted strings, but this will not allow to use both quoted and unquoted string in same type.

pabigot, 2020-06-15:
See also #21273 which relates to this request from the devicetree side.

Alternatively it could be a variant of general support for exposing property values that aren't in devicetree, one specific case being #25945.

mbolivar-nordic, 2022-08-18:
This was implemented in f4db14f349b2792445a13e12fb844731d8ed5c00.

closed 2022-08-18 by mbolivar-nordic