zephyrproject-rtos/zephyr #26209: sanitycheck tries to run random *samples*, without being asked for

pfalcon, 2020-06-16:
**Describe the bug**
Recently, a `sample.yaml` was added for samples/posix/eventfd:  https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/posix/eventfd/sample.yaml . The purpose of adding that file is to make sure that this sample is *built* in CI. As can be seen, the sample.yaml doesn't contain a directive telling "run this sample on the real hardware". Neither it contains directive like "run this sample on the real hardware and expect a particular reaction from hardware".

However, there's a report, #26185, saying that sanitycheck actually tries to run this particular sample on the hardware, and afterwards hangs, then times out - as if it waited some response from the hardware. But it's totally unclear what would it be waiting for. Perhaps, it waits for a camera to be attached, pointed to a board, to perform video recognition of an LED blinking on a board? That would be as wild guess as any other.

**To Reproduce**
See #26185.

**Expected behavior**
`sanitycheck` should not try to run random **samples**, unless explicitly instructed to do so in the sample.yaml. It is unsafe to run arbitrary samples, as it can be a sample of how to start nuclear meltdown, burn flash memory by repetitive erasing, or how to to start a DoS attach against servers on the Internet. Likewise, if sanitycheck is to expect any response from a sample, that to be stated explicitly in the sample.yaml. If it's not stated, sanitycheck should not expect any specific behavior from a sample. In particular, it should not stop and wait for something, unknown for a sample author/user.

Note that the discussion here applies exactly to **samples**, and not **tests**. Because tests indeed follow explicit conventions and intended to be used, well, for testing. Whereas **samples** are free-form and have very different purposes, and their usage for "testing" is strictly optional, and should not go without explicit specification of how a sample can be used for "testing". 

**Impact**
If this issue is not fixed, but some magic, implicit, unknown rules will be used for sanitycheck behavior regarding **samples**, cases like this will happen again and again, causing gross confusion and frustration among users.

**Screenshots or console output**
See #26185.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain  Zephyr SDK
 - Commit SHA or Version used: master cf72fe8fe9878a5c3ccf8824ea032d861d72720a

nashif, 2020-06-16:
> As can be seen, the sample.yaml doesn't contain a directive telling "run this sample on the real hardware".

All tests by default run on the platform they are built for, if that is possible, whether in emulation or real devices. This is the default policy and had been such for years now.

If you want a sample to only build, something we try to avoid, then you have to add `build_only: true` and the yaml file you reference clearly does not have that.

> However, there's a report, #26185, saying that sanitycheck actually tries to run this particular sample on the hardware, and afterwards hangs, then times out

Yes, this is the correct behavior, we do not only build, we also run, unless otherwise specified. build_only is one option, `harness` that is defined to something other than 'console`, etc. Way too many option to deal with this, but the one that always existed was 'build_only'

> Note that the discussion here applies exactly to **samples**, and not **tests**.

we also want to verify sample output, building the samples is not enough, wherever possible and if something can be run, it should be run and pass or failure can be determined based on the output, look at hello_world.

So, no bug here..

closed 2020-06-16 by nashif

pfalcon, 2020-06-18:
> All tests by default run on the platform they are built for

Sure, **tests** do. But here, we're talking about **samples**, and for that fact to not go unnoticed, there're all the `*` and bold emphasis.

> If you want a sample to only build, something we try to avoid, then you have to add build_only: true and the yaml file you reference clearly does not have that.

Not that I don't want it to be "build-only". A suitable testing agent is certainly welcome to run it. But that likely applies to a human. `sanitycheck` should not attempt to run samples which it doesn't know how to run (wasn't told how to run them). Or maybe it could, but it certainly should not hang on them, if it doesn't know what to expect from them (and unless it was told what to expect in `sample.yaml`, it doesn't know).

 > Yes, this is the correct behavior, we do not only build, we also run, 

... and you also hang, and you shouldn't do that. Unless explicitly asked to hang.

> build_only is one option, harness that is defined to something other than 'console`, etc. Way too many option to deal with this

Yes, that's the problem, too many options, mostly half-baked, then oftentimes applied in half-baked ways to actual tests/samples. But the issue we're talking about here is that **defaults are contra-logical** for *sample*.

> we also want to verify sample output, building the samples is not enough, wherever possible and if something can be run, it should be run and pass or failure can be determined based on the output, look at hello_world.

Sure, we also want Zephyr to fly to Mars, as legends says it did. In the meantime, what you do is making Zephyr testsuite runner to hang on absolutely innocent tests. At the very least, you should not do that. Maybe you want to further your agenda of making all samples be tested, and warn when there're no explicit test instruction (harness). But you definitely **should not hang when there're no such test instructions**.

> So, no bug here..

Sure, if the project owners want to collect tolls of this confusion in man-years, growth from current man-hours, ok. In the meantime, I hope people will make their conclusions, that no sample.yaml == no problems. (Writing less samples may also help.)

nashif, 2020-06-19:
> Sure, **tests** do. But here, we're talking about **samples**, and for that fact to not go unnoticed, there're all the `*` and bold emphasis.

samples have tests, hence, they are treated as tests. Just putting a sample in the tree and not do anything to verify it actually runs is something of the past, we have been there and right now we run anything that can be run on actual hardware and emulation and verify the output.

> Not that I don't want it to be "build-only". A suitable testing agent is certainly welcome to run it. But that likely applies to a human. `sanitycheck` should not attempt to run samples which it [...]

In this case, you mark the sample correctly using the many options available and just avoid running it because it requires more than what an emulator or a device with not enough hardware support provide..

> ... and you also hang, and you shouldn't do that. Unless explicitly asked to hang.

we timeout (hang) if the sample is not giving us what we are expecting, not sure how else you would want to do that, I expect XYZ, I am not getting the XYZ, so I wait for it until I timeout..

> Yes, that's the problem, too many options, mostly half-baked, then oftentimes applied in half-baked ways to actual tests/samples. But the issue we're talking about here is that **defaults are contra-logical** for _sample_.

the only thing contra-logicial here is your attitude to be honest.