zephyrproject-rtos/zephyr #26214: add support for armv8m in samples/*/code_recocation

hakonfam, 2020-06-16:
Currently the `code_relocation` sample use registers not present in armv8 devices when `CONFIG_ARM_MPU` is enabled:
```
	for (index = 0U; index < 8; index++) {
		MPU->RNR = index;
		if (MPU->RASR & MPU_RASR_XN_Msk) {
			MPU->RASR ^= MPU_RASR_XN_Msk;
		}
	}
```

`RASR` is replaced by `RBAR` and `RLAR` in v8m, hence the sample will not compile for these architectures.

Either use more high level functionality from `arm_mpu.c` and friends to be more generic, or add a separate handling for other arches.

Also the sample should not build for non-supported devices (has MPU but does not have proper handling).

KevinThompson-SR, 2020-09-24:
Curious if issues have been reported regarding the gen_relocate_app.py?
[image]

hakonfam, 2020-09-25:
Not sure I understand what you mean, but I have not reported any issues regarding that at least.

KevinThompson-SR, 2020-09-25:
zephyr\samples\application_development\code_relocation will not build in release 2.3, at least when targeting nrf5340.  See gen_relocate_app.py output above.  I think my build environment is, OK, any pointers to address this issue would be greatly appreciated.

closed 2025-08-29 by wearyzen