zephyrproject-rtos/zephyr #26225: x86_64 doesn't seem to be handling spurious interrupts properly

andrewboie, 2020-06-16:
**Describe the bug**

>  The last issue was that there is apparently stray interrupt being signaled by... something, somewhere between OS entry and when we try to swap to main (which enables interrupts).  It turns out x86_64 doesn't have a spurious interrupt handler, so on interrupt it just jumps to NULL.

This shouldn't be happening, we ought to be getting a fatal error with reason K_ERR_SPURIOUS_IRQ.

Need to check x86_64's code path on spurious interrupts.

andrewboie, 2020-06-16:
If it matters, this was an IRQ on vector 32 (which I believe Zephyr reserves)

ceolin, 2020-10-22:
Commit c7d76cbe5844fb96823ae6467d8f10a8d3593cc1 installed a spurious handlers that triggers a fatal error with `K_ERR_SPURIOUS_IRQ` just like in others platforms. This commit was done after this pr be opened.

closed 2020-10-22 by ceolin