zephyrproject-rtos/zephyr #26232: Segger Embedded Studio doesn't find the right python

sandeep-airatechnology, 2020-06-17:
When I try to build Zephyr in Segger Embedded Studio, I see a missing 'elftools' error.
The terminal way of building works using 'west build -p auto -b nrf52840_pca10056 samples/basic/blinky'

Error seen:
[image]

However pyelftools is installed.
Looking closely I noticed that the python SES is picking up is in /usr/bin/python3. 
pyelftools is installed in /usr/local/bin/python. 'which python' in bash terminal points to the said path.
How do I make SES choose the right python?

stephanosio, 2020-06-17:
`/usr/bin/python3` would be the system default Python (included in macOS), and `/usr/local/bin/python` would likely be the one you installed through Homebrew.

This looks more like a problem with the SES, not Zephyr.

SandeepKesiReddy, 2020-06-17:
I checked that.
'which python3' points to /usr/local/bin/python3.

Is there a way to point SES to right path, does it look at any project config files?

sandeep-airatechnology, 2020-06-17:
I checked that.
'which python3' points to /usr/local/bin/python3.

Is there a way to point SES to right path, does it look at any project config files?

tejlmand, 2020-06-18:
This is unfortunately a macOS issue.

When launching an app (such as SES) through command line, it will inherit the `$PATH` from the shell (and then the right Python should be found).
But if it is launched by just clicking the `application` (SES) icon, then `/etc/paths` are not considered in macOS, which causes SES (in fact CMake) to not find the expected Python.

You can set the exact path to the Python you want to use, by going through: `Tools` --> `Options` in SES:
[image]

Note: we have discovered an issue where `python` and `west-python` can be out of sync, and based on this issue, it could happen to you, so also read: #26060

sandeep-airatechnology, 2020-06-19:
Thank-you.. This solved the problem..
I can now build and debug with SES..

closed 2020-06-19 by sandeep-airatechnology