zephyrproject-rtos/zephyr #26235: multi vlan support networking

eyuval, 2020-06-17:
For my current project I need the following features:
1. A network port can be part of several vlans. 
2. Applications should be able to specify, which vlan they wish to work with, as in linux, where a virtual network device is created and the application can choose to work with e.g.
 eth0     -- without vlan
 eth0.7  -- with vlan7
 eth0.8  -- with vlan8

jukkar, 2020-06-17:
> * A network port can be part of several vlans.

By "port" do you mean network interface?

> 
> * Applications should be able to specify, which vlan they wish to work with, as in linux, where a virtual network device is created and the application can choose to work with e.g.
>   eth0     -- without vlan
>   eth0.7  -- with vlan7
>   eth0.8  -- with vlan8

VLANs are configured similar way in zephyr too. After starting the application, it needs to call `net_eth_vlan_enable(iface, vlan_tag);` to attach VLAN tag to the interface and enable VLAN.

jukkar, 2020-06-17:
Yes, this is exactly how zephyr works too. One bigger difference is that the amount of VLANs need to be known in advance as the virtual VLAN interfaces are allocated during compile time. At runtime application will need to enable the actual VLAN functionality. See how to do that for example in `samples/net/sockets/echo_server/src/vlan.c`

eyuval, 2020-06-17:
Thanks for your reply. I had a look at the sample you mention, but it seems to me, that it only adds 1 vlan per interface. For me the problem manifests itself when I call the shell commands:
net vlan add 7 1
net vlan add 8 1
Here I get an error for the second command. Currently our net_if interfaces reflect our physical network ports, so their number is the same as the number of physical network ports. I understand your explanation such, that we should create more interfaces than our physical ports and do the vlan add on these 'virtual interfaces'.

jukkar, 2020-06-18:
>  it only adds 1 vlan per interface

Yes, but that interface is the "VLAN" virtual interface. So in Zephyr, there is the master interface (like eth0 in Linux), then there are virtual VLAN interfaces (like eth0.10 etc in Linux). All of these interfaces are created in advance during compilation. The virtual interfaces are not usable themselves for VLAN, but require setting of VLAN tag and IP addresses etc. See also documentation at https://docs.zephyrproject.org/latest/reference/networking/vlan.html

So in the net-shell example, you should not try to add VLAN to the master interface but to the VLAN interfaces so something like
net vlan add 7 2
net vlan add 8 3

eyuval, 2020-06-18:
Thanks for the explanation. Now it is much clearer.

ehudmarvell, 2020-06-18:
@jukkar 
Thanks for you answer.
I will try to ask specific question.
Currently, we are creating 1 interface per physical port by using:
`NET_DEVICE_INIT_INSTANCE(...)`

If I would like to create in addition virtual interface, I will create more instance by using?
`NET_DEVICE_INIT_INSTANCE`

if yes, so now I have 2 interface per one physical port, 
for example RX side on this port, I am  receiving packets by using 
`if (net_recv_data(iface, pkt) < 0) {`

should I have one iface for both of the interface? or I need to parsing it by vlan id and select the right interface?
If they have the same iface, what is the MACRO do it?

Thanks a lot,
Ehud
 
  

Thanks,
Ehud

jukkar, 2020-06-18:
> Currently, we are creating 1 interface per physical port by using:
> `NET_DEVICE_INIT_INSTANCE(...)`

You need to use `ETH_NET_DEVICE_INIT()` to create all the virtual VLAN interfaces, the `NET_DEVICE_INIT()` does not support VLANs.

ehudmarvell, 2020-06-22:
Hi @jukkar 

Thank you for your help,
another question, I found  an issue:
1. I have 2 interface on the same physical port.
2. I would like that both of them have the same mac address -> and also the same ipv6 local address.
3. I configure the second interface to be with vlan, while the first interface without it.
4. while ping to the vlan interface the packet is dropped in tx.

It is drop in  ethernet_send->set_vlan_tag
 ```
tatic enum net_verdict set_vlan_tag(struct ethernet_context *ctx,
				     struct net_if *iface,
				     struct net_pkt *pkt)
{
	int i;

	if (net_pkt_vlan_tag(pkt) != NET_VLAN_TAG_UNSPEC) {
		return NET_OK;
	}

#if defined(CONFIG_NET_IPV6)
	if (net_pkt_family(pkt) == AF_INET6) {
		struct net_if *target;

		if (net_if_ipv6_addr_lookup(&NET_IPV6_HDR(pkt)->src,
					    &target)) {
			if (target != iface) {
				NET_DBG("Iface %p should be %p", iface,
					target);

				iface = target;
			}
		}
	}
```
the net_if_ipv6_addr_lookup will return the first interface ptr instead the second, as result afterwards it wont find the vlan(becasue it isn't exist on the first iface) and the packet will be drop.

I am suggesting to replace the 
```

if (net_if_ipv6_addr_lookup(&NET_IPV6_HDR(pkt)->src,
					    &target)) {
			if (target != iface) {
				NET_DBG("Iface %p should be %p", iface,
					target);

				iface = target;
			}
		}
```
with
```
               if (!net_if_ipv6_addr_lookup_by_iface(iface, &NET_IPV6_HDR(pkt)->src)) {
                      return NET_DROP;
                }
```
of course, similar thing for Ipv4 too, first I would like to understand what do you think.

Thanks,
Ehud

jukkar, 2020-07-24:
Hi @ehudmarvell, sorry for late reply, I just got back from vacation.

> * have 2 interface on the same physical port.
> 
> * I would like that both of them have the same mac address -> and also the same ipv6 local address.

The MAC address will be the same for VLAN ports as there is only one physical network interface involved. Each port will have unique IPv6 link local address, the reason being that the neighbor cache implementation in zephyr uses IPv6 address to find out the neighbor, so cache could return wrong network interface in that case.

I am not sure what exactly is the issue. Could you prepare separate issue instead of adding new stuff to this one. If you have patches, just them as PRs, it is easier to review them that way and also try and test them.

jukkar, 2020-07-24:
I am closing this one as the original question was answered.

closed 2020-07-24 by jukkar