zephyrproject-rtos/zephyr #26242: qemu_x86 time handling broken

carlescufi, 2020-06-17:
```
        k_sem_init(&sync_sem, 0, 1);
[...]
        printk("uptime 1: %lld\n", k_uptime_get());
        err = k_sem_take(&sync_sem, K_SEC(10));
        printk("uptime 2: %lld\n", k_uptime_get());
        BT_ASSERT_MSG(err == 0, "k_sem_take failed with err %d", err);
```

The code executes immediately, and the result is:

```
$ time west build -t run
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200516_152813-e7ee184eb100-zephyr
)
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 7
*** Booting Zephyr OS build zephyr-v2.3.0-415-g3f1991804144  ***
Starting Beacon Demo
uptime 1: 0
uptime 2: 800
uptime 1: 800
uptime 2: 10810
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:372
        k_sem_take failed with err -11
[00:00:10.810,000] <err> os: EAX: 0x00109d59, EBX: 0x00109d59, ECX: 0x0000000a, EDX: 0x0011c72c
[00:00:10.810,000] <err> os: ESI: 0xfffffff5, EDI: 0x0000000a, EBP: 0x0011c764, ESP: 0x0011c72c
[00:00:10.810,000] <err> os: EFLAGS: 0x00000206 CS: 0x0008 CR3: 0x0011a180
[00:00:10.810,000] <err> os: call trace:
[00:00:10.810,000] <err> os: EIP: 0x00105092
[00:00:10.810,000] <err> os:      0x00105bd5 (0x1003)
[00:00:10.810,000] <err> os:      0x00105fe4 (0x1)
[00:00:10.810,000] <err> os:      0x00100ef9 (0x11d108)
[00:00:10.810,000] <err> os:      0x00100eb4 (0x11a4e0)
[00:00:10.810,000] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:10.810,000] <err> os: Current thread: 0x0011a500 (unknown)
[00:00:10.810,000] <err> os: Halting system
FAILED: zephyr/CMakeFiles/run
cd /home/carles/src/zephyr/zephyr/build && /home/carles/bin/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-i386 -m 9 -cpu qemu32,+nx,+pae -device isa-debug-exit,iobase=0xf4,iosize=0x04 -no-reboot -nographic -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -serial unix:/tmp/bt-server-bredr -icount shift=5,align=off,sleep=off -rtc clock=vm -kernel /home/carles/src/zephyr/zephyr/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/carles/bin/cmake/cmake-3.13.1-Linux-x86_64/bin/cmake --build /home/carles/src/zephyr/zephyr/build --target run

real    0m0.425s
user    0m0.321s
sys     0m0.122s
```

wentongwu, 2020-06-17:
> but he simplest really is to build samples/philosophers for qemu_x86 and see how real time is not respected.

@carlescufi this is because qemu icount mode uses different time standard with wall time. 

> BT_ASSERT_MSG(err == 0, "k_sem_take failed with err %d", err);

haven't much experience about BT and how it works with QEMU, but as classic bt spec, it highly depends on timing to communicate with others, if master and slave uses different time standard(most like frequency drift), it will not work. So I suggest disable icount when BT protocol running on QEMU.

wentongwu, 2020-06-17:
icount mode can only be run on self involved cases, and not sure is there any other networking or wireless protocol supported by Zephyr but can't be covered by `default y if !NETWORKING || !BT`? 

```
diff --git a/boards/Kconfig b/boards/Kconfig
index 6e4af598f3..472193fb8d 100644
--- a/boards/Kconfig
+++ b/boards/Kconfig
@@ -31,7 +31,7 @@ menu "Board Options"
 config QEMU_ICOUNT
        bool "QEMU icount mode"
        depends on QEMU_TARGET
-       default y if !NETWORKING
+       default y if !NETWORKING || !BT
        help
          Enable QEMU virtual instruction counter. The virtual cpu will
          execute one instruction every 2^N ns of virtual time. This will
```

jhedberg, 2020-06-17:
@wentongwu only the Bluetooth controller has very strict timing requirements, and we don't run that in qemu (we only run the host-side there and the controller is connected externally)

carlescufi, 2020-06-17:
> icount mode can only be run on self involved cases, and not sure is there any other networking or wireless protocol supported by Zephyr but can't be covered by `default y if !NETWORKING || !BT`? [...]

This looks good enough, BT cannot work with time that is not wall time in practice, especially with these massive differences. I will send a PR.

PR #26251 (2020-06-18): boards: kconfig: Disable QEMU icount when Bluetooth is enabled
Bluetooth interacts with real-world hardware and thus requires the QEMU
target to follow wall time, and not have a free running timer that is
much faster than actual wall time.

Diable the QEMU icount mechanism when Bluetooth is enabled.

Fixes #26242
commit 24f8b588 (2020-06-18): boards: kconfig: Disable QEMU icount when Bluetooth is enabled

Bluetooth interacts with real-world hardware and thus requires the QEMU
target to follow wall time, and not have a free running timer that is
much faster than actual wall time.

Diable the QEMU icount mechanism when Bluetooth is enabled.

Fixes #26242
closed 2020-06-18 by carlescufi