zephyrproject-rtos/zephyr #26246: Printing 64-bit values in LOG_DBG

asbjornsabo, 2020-06-17:
**Describe the bug**
`LOG_DBG` does not seem to correctly output the value of uint64 variables when the value is larger than 32 bits.
I expect values of 2^^32 and larger to be printed as the value they are.
What happens is that the value 0 (zero) is printed instead.

For comparison, `printk` does print these values correctly, as does `printf` (when using NEWLIB).

**To Reproduce**
Steps to reproduce the behavior:

The code I made to reproduce this can be found here:
https://github.com/asbjornsabo/zephyr/tree/printf_64bit/samples/printf_64bit

The code in question is:
```
void main(void)
{
	uint64_t id = 1;
	uint8_t startshift = 30;
	uint8_t endshift = 34;
	uint8_t shift;

	id = id << startshift;

	for( shift = startshift; shift < endshift; shift++) {
		LOG_DBG("debug: The value of \"shift\" is %d, the value of \"id\" is 0x%llx\n", shift, id);
		printk("printk: The value of \"shift\" is %d, the value of \"id\" is 0x%llx\n", shift, id);
		printf("printf: The value of \"shift\" is %d, the value of \"id\" is 0x%llx\n", shift, id);
		id = id << 1;
	}
}
```
My prj.conf has
```
CONFIG_SERIAL=y
CONFIG_UART_NRFX=y
CONFIG_LOG=y
CONFIG_LOG_BACKEND_UART=y
CONFIG_LOG_DEFAULT_LEVEL=4
CONFIG_DEBUG=y

CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_NANO=n
```

What I do to reproduce is:
rm -rf build && cmake -H. -Bbuild -GNinja -DBOARD=nrf52840dk_nrf52840
cd build && ninja
(cd ..)
nrfjprog --eraseall --snr 683854931 && nrfjprog --program build/zephyr/zephyr.hex --verify --snr 683854931 && nrfjprog --reset --snr 683854931  
(Replace SNR with applicable value.)

The output I get is:
printk: The value of "shift" is 30, the value of "id" is 0x40000000
printf: The value of "shift" is 30, the value of "id" is 0x40000000
printk: The value of "shift" is 31, the value of "id" is 0x80000000
printf: The value of "shift" is 31, the value of "id" is 0x80000000
printk: The value of "shift" is 32, the value of "id" is 0x100000000
printf: The value of "shift" is 32, the value of "id" is 0x100000000
printk: The value of "shift" is 33, the value of "id" is 0x200000000
printf: The value of "shift" is 33, the value of "id" is 0x200000000
[00:00:00.004,608] <dbg> printf_64bit.main: debug: The value of "shift" is 30, the value of "id" is 0x40000000

[00:00:00.016,845] <dbg> printf_64bit.main: debug: The value of "shift" is 31, the value of "id" is 0x80000000

[00:00:00.029,022] <dbg> printf_64bit.main: debug: The value of "shift" is 32, the value of "id" is 0x0

[00:00:00.041,412] <dbg> printf_64bit.main: debug: The value of "shift" is 33, the value of "id" is 0x0

**Expected behavior**
The output I would expect is for LOG_DBG to output the same values as printk/printf.

**Impact**
As part of upcoming code for le-audio, I need to output 64 bit values for debugging.  I currently have self-made function for this.  This adds to the code, and I have been asked by the maintainer to replace this with more "standard" zephyr solutions.

**Screenshots or console output**
(See pasted otuput above.)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.11.3
 - Commit SHA or Version used: Branched from master at 9e2bdaa3325238919c28c8bfad6db3ba9ba85

**Additional context**
Add any other context about the problem here.

carlescufi, 2020-06-17:
@asbjornsabo I believe that the safe identifier to use is `PRIx64`

jukkar, 2020-06-17:
I have created similar enhancement req in the past #12098

andrewboie, 2020-06-17:
The log mechanism uses the printk code (`snprintk()`, `vsnprintk()`, etc.) to manage format strings right? That code would need to be extended in order to support 64-bit integers on 32-bit systems, I don't believe it does currently.

The caveat is that printk() code is intentionally kept small since it's part of the core kernel, we want to be very careful about increasing its footprint (either code size or stack usage). Parts of the printf spec have been deliberately left out of it.

One possible alternative might be a config option for the log subsystem code to use the C library's `vsnprintf()` or `snprintf()` instead, which is fuller-featured (at the expense of code size and stack usage).

@jukkar correctly points out that the minimal libc printf() functions don't do this right either, but I don't see anything controversial about adding it there.

asbjornsabo, 2020-06-18:
> @asbjornsabo I believe that the safe identifier to use is `PRIx64`
@carlescufi Plese explain what you mean by "safe identifier" and where I should use it.  (I assume you are not meaning as the format specifier?)

asbjornsabo, 2020-06-18:
> The log mechanism uses the printk code (`snprintk()`, `vsnprintk()`, etc.) to manage format strings right? That code would need to be extended in order to support 64-bit integers on 32-bit systems, I don't believe it does currently.
> [...]
@andrewboie If the log mechanism uses printk code (I think I may have seen a vsnprintk or similar in there), why does `printk()` print the value correctly, and `LOG_DBG` not?  
(See the output - printk works.)

nordic-krch, 2020-06-18:
> @andrewboie If the log mechanism uses printk code (I think I may have seen a vsnprintk or similar in there), why does printk() print the value correctly, and LOG_DBG not?

that is the limitation of the logger. It stores each argument in `log_arg_t` which is `typedef unsigned long log_arg_t;` and  has 32 bit on 32 bit platforms. I need to check if storing on 64bits would solve the issue. If yes then i can add such option to the logger but it won't the the default since it will lead to increased log message size (more ram needed for log messages).

nordic-krch, 2020-06-22:
I tried to check if it is possible to redefine `log_arg_t` to `uint64_t` on 32bit platforms but i hit the issue with compilation warning when casting. Each log argument is casted to `log_arg_t` https://github.com/zephyrproject-rtos/zephyr/blob/ad59e923e948605adf3baa44361cf1da2c22be50/include/logging/log_core.h#L201
and compiler complains when `void *` is casted to 64 bits. I don't know how to do that cast then. If anyone can suggest something?

andrewboie, 2020-06-22:
`log_arg_t` I have seen lead to some very cryptic and misleading warning messages in existing code. What mechanism is this intended to support?

asbjornsabo, 2020-08-28:
Are there any plans for progress on this?

pabigot, 2020-09-14:
Note that any solution to this would handle #18351 as well.  It's not a simple problem to solve; some history is in https://github.com/zephyrproject-rtos/zephyr/issues/18351#issuecomment-522276594

asbjornsabo, 2021-02-17:
I am still interested in a solution to this.

nordic-krch, 2021-02-17:
It's fixed in logging overhaul.

closed 2021-04-19 by nashif

nordic-krch, 2021-04-20:
64 bit values can now be logged with new logging. Currently legacy version is the default one, so it has to be explicitly enabled (`CONFIG_LOG2_MODE_DEFERRED`).

asbjornsabo, 2021-04-20:
Confirmed - it works now.