zephyrproject-rtos/zephyr #26252: bluetooth: controller: DLE calculations are not correct

rugeGerritsen, 2020-06-17:
**Describe the bug**
When running the NCS throughput sample with SoftDevice LL as master and Zephyr LL as slave, the connection drops after the master starts transmitting long packets.

**To Reproduce**
Steps to reproduce the behavior:
Checkout NCS v1.3.0-rc3
Run throughput sample on two nrf52 devkits.
Configure the SoftDevice LL to be master, Zephyr LL to be slave

**Expected behavior**
The Zephyr LL should be able to receive long packets.

**Impact**
Showstopper. It annoys the user because it initially looks like it is the SoftDevice LL which is misbehaving. 

**Screenshots or console output**
Entire sequence:
[image]

The screenshot below shows pin toggles on RADIO_READY and RADIO_DISABLED for both the master and slave. You can see that the Zephyr LL toggles multiple times during the packet sent from the master.
[image]

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: gnuarmemb, 8 2019 q3
 - Commit SHA or Version used: NCS v1.3.0-rc3

rugeGerritsen, 2020-06-17:
@cvinayak , @kruithofa FYI

cvinayak, 2020-06-18:
@rugeGerritsen if you have the sniffer logs, upload here if the size is permitted, else by other means.

rugeGerritsen, 2020-06-18:
@cvinayak. Here are two sniffer logs. One for the default configuration and one for the case where only 1M PHY is used
[ThroughputSampleFails.zip](https://github.com/zephyrproject-rtos/zephyr/files/4796669/ThroughputSampleFails.zip)

cvinayak, 2020-06-18:
I have identified the bug in Zephyr LL control procedure handling of unknown rsp. I will send a fix as soon as I can get back to work.

PR #26300 (2020-06-22): Bluetooth: controller: Add explicit opcode check in unknown rsp PDU
Add explicit opcode check when handling received unknown
response PDU.

Without this, for example, an in progress Data Length Update
procedure state was reset when receiving an unknown response
to slave initiated feature request.

Fixes #26252.
commit 1af8c136 (2020-06-22): Bluetooth: controller: Add explicit opcode check in unknown rsp PDU

Add explicit opcode check when handling received unknown
response PDU.

Without this, for example, an in progress Data Length Update
procedure state was reset when receiving an unknown response
to slave initiated feature request.

Fixes #26252.
PR #26404 (2020-09-15): [backport v2.2] Bluetooth: controller: Add explicit opcode check in unknown rsp PDU
Add explicit opcode check when handling received unknown
response PDU.

Without this, for example, an in progress Data Length Update
procedure state was reset when receiving an unknown response
to slave initiated feature request.

Fixes #26252.
commit c895d1d0 (2020-09-15): Bluetooth: controller: Add explicit opcode check in unknown rsp PDU

Add explicit opcode check when handling received unknown
response PDU.

Without this, for example, an in progress Data Length Update
procedure state was reset when receiving an unknown response
to slave initiated feature request.

Fixes #26252.
closed 2020-06-22 by carlescufi