zephyrproject-rtos/zephyr #26255: k_uptime_ticks() returns pointer instead of value

govindadas, 2020-06-17:
**Describe the bug**
k_uptime_ticks() should return a value but it behaves like a pointer when used on a nRF52dk board.
The definition from kernel.h:
> @brief Get system uptime, in system ticks.
>   This routine returns the elapsed time since the system booted, in
>   ticks (c.f. :option:`CONFIG_SYS_CLOCK_TICKS_PER_SEC`), which is the
>   fundamental unit of resolution of kernel timekeeping.
>   @return Current uptime in ticks.
> __syscall s64_t k_uptime_ticks(void);

I'm not sure if it's a Bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Open samples/bluetooth/central_hr project
2. Make sure output is enabled
3. Add this code to main():
```
 while(1){
      s64_t ticks1 = k_uptime_ticks();
      printk(" %u ", ticks1);
      s64_t *ticks2 = k_uptime_ticks();
      printk(" %u \n", ticks2);
      k_sleep(K_MSEC(10));
 }
```
4. Flash to board: nrf52dk_nrf52832
5. See printk() output for error: 
`0  7029`
First printk is 0, second printk is real ticks value. 

**Expected behavior**
First printk should be the real ticks value

**Impact**
annoyance, can't use k_uptime_get() because it uses k_uptime_ticks()

**Screenshots or console output**
```
0  7029 
0  7360 
0  7691 
0  8022 
```

**Environment:**
 - OS: Linux
 - Toolchain Zephyr SDK, nRF Connect SDK, GNU Arm Embedded Toolchain
 - Zephyr 2.3.0 and 2.2.0
 - Board: nrf52dk_nrf52832

carlescufi, 2020-06-17:
I think you are just not printing your 64-bit values correctly. Try using `%llx` instead of `%u`.

govindadas, 2020-06-17:
> I think you are just not printing your 64-bit values correctly. Try using `%llx` instead of `%u`.

Yes, you are right! Stupid mistake. 
@carlescufi Thank you!
Sorry for wasting everyone's time...

closed 2020-06-17 by govindadas