zephyrproject-rtos/zephyr #26256: NRF51822 BLE Micro module: hangs on k_msleep() (RTC counter not working)

v01d, 2020-06-17:
**Describe the bug**
I'm trying to run Zephyr on Seeedstudio BLE Micro module, based on NRF51822 (16KB RAM VERSION), featuring both 16MHz and 32KHz crystal. Initially I defined an out of tree board based on the BLE Nano board, modifying required definitions. I tried the blinky example and while the LED turns on, it does not blink. Debugging via SWD shows that the board hangs on WFI instruction and only wakes if I pause execution with debugger. Further inspection appears to indicate that the RTC counter used for Zephyr sleep timings does not produce interrupts on timer expiration nor the counter appears to increase.
I decided to go back and simply modify the in-tree board for BLE Nano, just a few simple modifications needed, and build the same example. The result was the same. I also went into menuconfig and chose alternative source of system clock timer, which by default was set to internal RC. Same thing happens If I switch to external oscillator (32KHz) or derived from HFCLK.
It should be considered that when flashing the nrfUART example firmware I can see the BLE device and pair with it (although can't test more since those UART pins are not mapped in my PCB).

There's a chance that this might be defective hardware since I had to desolder the module with hot air and resolder it. However, I'm reluctant to consider that since the other firmware works and the it would be strange that the internal RC oscillator does not work, while the board is able to function normally up to that point.

**To Reproduce**
Steps to reproduce the behavior:
1. download zephyr repo, set release to 2.3.0
2. go to blinky directory
3. west build -b nrf51_blenano
4. Flash image

To flash I'm using ST Link v2 clone device and following command (since west flash complained about my device not being up to date):
openocd -f interface/stlink-v2.cfg -f target/nrf51.cfg -c init -c "reset halt" -c "nrf51 mass_erase" -c "program build/zephyr/zephyr.hex verify reset" -c exit

**Expected behavior**
The LED should blink 

**Impact**
Cannot do anything else if this does not work

**Screenshots or console output**
Nothing to show

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: v2.3.0 tag

**Additional context**

v01d, 2020-06-19:
any suggestion on what I could try?

cvinayak, 2020-06-19:
I have not used nRF51 DK (has the 32KHz crystal) or BBC microbit (does not have 32KHz crystal) recently. Give me time until Monday.

v01d, 2020-06-19:
sure, thanks

cvinayak, 2020-06-19:
@v01d I have tested `bbc_microbit` board with `samples/bluetooth/scan_adv` which uses nRF51 with internal RC for LFCLOCK. The sample works as expected. Could you build the same with no local changes (ignore UART mapping etc.) and check if Bluetooth advertising is seen from your module.
Also, please attach your `.config` file.

v01d, 2020-06-19:
I just found the problem. It is a silicon errata not considered in Zephyr. The errata in question is here:
https://infocenter.nordicsemi.com/pdf/nRF51822-pan_v2.1.pdf (page 26)
Since a LED blink example worked when using platformio, based on mBed framework, I looked at the code and they include the suggested workaround:

https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.15/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/device/system_nrf51.c#L93

I just introduced this code right before starting the RTC peripheral and the LED blinks.

v01d, 2020-06-19:
I continued testing and while I'm now able to run the blink example and I also get UART output (if the workaround is placed right after system init, so that it runs before any peripheral is configured), I don't seem to be getting Bluetooth connectivity. I cannot see the device from my phone nor the device can see my phone.
I'm not sure if there are further problems for this chip revision or something else is wrong.

closed 2020-09-13 by github-actions[bot]

joerchan, 2021-08-25:
> I just found the problem. It is a silicon errata not considered in Zephyr. The errata in question is here:
> https://infocenter.nordicsemi.com/pdf/nRF51822-pan_v2.1.pdf (page 26)
> Since a LED blink [...]

The errata is not included due to the required defines not being defined. Should be fixed by this: https://github.com/zephyrproject-rtos/zephyr/pull/37925