zephyrproject-rtos/zephyr #26263: tests/posix/common failed on reel_board.

chen-png, 2020-06-18:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b reel_board -p auto tests/posix/common
2. west flash
3. See error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.3.0-397-g540ed9578613  ***
Running test suite posix_apis

starting test - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15
Bounce test OK
Barrier test OK
PASS - test_posix_pthread_execution

starting test - test_posix_pthread_termination
Thread 0 starting with a priority of 4
Cancelling thread 0
Thread 1 starting with a priority of 3
Cancelling thread 1
Thread 1 could not be cancelled
Thread 2 starting with a priority of 2
Cancelling thread 2
Thread 3 starting with a priority of 1
Cancelling thread 3
Thread 3 could not be cancelled
PASS - test_posix_pthread_termination

starting test - test_posix_multiple_threads_single_key
Different threads set different values to same key:
thread 0: set value = 536877516 and retrieved value = 536877516
thread 1: set value = 536877756 and retrieved value = 536877756
PASS - test_posix_multiple_threads_single_key
....
starting test - test_nanosleep_0_1000000000
PASS - test_nanosleep_0_1000000000

starting test - test_nanosleep_0_1
dt_ns: 30518
PASS - test_nanosleep_0_1

**starting test - test_nanosleep_0_500000000
dt_ns: 499847413
Assertion failed at WEST_TOPDIR/zephyr/tests/posix/common/src/nanosleep.c:203: common: (dt_ns >= ns is false)
expected dt_ns >= 500000000: actual: 499847413**
FAIL - test_nanosleep_0_500000000

starting test - test_nanosleep_1_0
dt_ns: 999908448
Assertion failed at WEST_TOPDIR/zephyr/tests/posix/common/src/nanosleep.c:203: common: (dt_ns >= ns is false)
expected dt_ns >= 1000000000: actual: 999908448
FAIL - test_nanosleep_1_0

Test suite posix_apis failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.11.3
 - Commit ID: 540ed95786

PerMac, 2020-06-18:
I see these fails (test_nanosleep_0_500000000 and test_nanosleep_1_0) also on nrf52dk_nrf52832, nrf52840dk_nrf52840 and nrf9160dk_nrf9160.

ioannisg, 2020-07-21:
The test failure has something to do with the nRF RTC driver and the system timer configurations (cycles, ticks, etc.). The test does not fail on nRF platforms if the SysTick timer is used instead of the RTC.

Assigning this to nRF timer experts @pabigot , @nordic-krch

pabigot, 2020-07-21:
https://github.com/zephyrproject-rtos/zephyr/issues/24001#issuecomment-633529834 may be relevant.  @jfischer-phytec-iot any updates?

pabigot, 2020-07-22:
This is the same problem that we hacked around for `kernel/timer/timer_api` and `portability/cmsis_rtos_v1`:  you can't reliably compare times calculated with two different clocks (in this case Nordic cycle timer from LFCLK and busywait from the CPU HFCLK).  I'll add code to the new tests to match those.

PR #27065 (2020-09-14): tests: posix: fix nanosleep tests for clock slew
Use the technique from portability/cmsis_rtos_v1 to compensate for
systems where times are being measured with different clocks.

Fixes #26263.
closed 2020-09-14 by carlescufi