zephyrproject-rtos/zephyr #26265: Zephyr os bluetooth peripheral example indication. When i flash vode to my board custom configuration for indication will shown and after i click button for indication it device will disconnect from phone. My board is nrf52832.

athulvalsanvp, 2020-06-18:
I am using custom characteristic for indication. I am attaching screen shot for your reference.

[image]
[image]

cvinayak, 2020-06-19:
Please provide board details, sample or the modified application used, build details, how to reproduce, screenshots and console logs (preferrably please use the github issue template that was presented when you created this issue).

athulvalsanvp, 2020-06-19:
[image]
[image]
Board: nrf52832.
Modified Example: bluetooth/peripheral
From center device(phone) i can see indication characteristics . When i click button for indication. It shows Device disconnected and also shows error 22

carlescufi, 2020-07-10:
Please send an email to the mailing list of contact us on Slack for code-related questions.

closed 2020-07-10 by carlescufi