zephyrproject-rtos/zephyr #26278: bt_att: Unhandled ATT code 0x1d

mfiumara, 2020-06-18:
**Describe the bug**
Local device performing ATT write cmds, generates <wrn> bt_att: Ignoring unexpected request when peer device send an ATT request. The peer then gets <err> bt_att: ATT timeout

To Reproduce use the tag 2.2.0

**Steps to reproduce the behavior**
Build and flash peripheral:

mkdir -p build/peripheral; cd build/peripheral
cmake -GNinja -DBOARD=nrf52dk_nrf52832 ../../samples/bluetooth/peripheral
ninja
ninja flash

Build and flash central_hr:
5. mkdir -p build/central_hr; cd build/central_hr
6. cmake -GNinja -DBOARD=nrf52dk_nrf52832 ../../samples/bluetooth/central_hr
7. ninja
8. ninja flash

Open two terminals, minicom -D /dev/ttyACMx
Observe both peripheral and central_hr fail with respective warning and error messages in the console.

**Expected behavior**
ATT request from peer device shall not be dropped. ATT bearer shall not timeout.

**Impact**
showstopper

Screenshots or console output
```
[00:24:57.234,680] <wrn> bt_ctlr_hci: type: 0x14
[00:24:57.332,244] <wrn> bt_att: Unhandled ATT code 0x1d
[00:24:57.430,297] <wrn> bt_att: Ignoring unexpected request
```
image

**Environment (please complete the following information)**

- OS: Linux
- Toolchain: gnu arm embedded
- Commit SHA or Version used: 2.2.0

Additional context
Add any other context about the problem here.

carlescufi, 2020-06-22:
@mfiumara could you please try to reproduce this in master?

Vudentz, 2020-06-22:
Interesting, 2.2 did not contain EATT changes so I wonder how this issue appear there as well, perhaps it is a different problem.

mfiumara, 2020-06-24:
@carlescufi I can't seem to reproduce this even on `2.2.0`. I will let you know if I encounter this again but I have seen it on 2.2.0.

carlescufi, 2020-07-10:
Closing for now, please reopen if you see it again. Thanks!

closed 2020-07-10 by carlescufi