zephyrproject-rtos/zephyr #26280: test_kernel_systicks from tests/portability/cmsis_rtos_v1 fails on nrf platforms

PerMac, 2020-06-18:
**Describe the bug**
The test case _test_kernel_systicks_ from tests/portability/cmsis_rtos_v1 test suite fails consistently at nrf platforms: (nrf52dk_nrf52832, nrf52840dk_nrf52840 and nrf9160dk_nrf9160)

**To Reproduce**
Steps to reproduce the behavior:
1. run in zephyr dir ./scripts/sanitycheck --testcase-root ./tests/portability/cmsis_rtos_v1 --device-testing -p nrf52dk_nrf52832 -v -v --device-serial /dev/ttyACM0
2. See error

**Expected behavior**
The test case test_kernel_systicks passes on nrf platforms

**Impact**
not clear

**Screenshots or console output**
```
DEBUG   - DEVICE: Assertion failed at CMAKE_SOURCE_DIR/src/kernel_apis.c:58: test_kernel_systick: (diff <= max && diff >= min is false)
DEBUG   - DEVICE: start 991 stop 34848 (diff 1033233) wait 1000000
DEBUG   - DEVICE: 
DEBUG   - DEVICE: FAIL - test_kernel_systick
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: starting test - test_thread_apis
DEBUG   - DEVICE: PASS - test_thread_apis
DEBUG   - DEVICE: ===================================================================
[...]
DEBUG   - DEVICE: Test suite test_cmsis_apis failed.
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: PROJECT EXECUTION FAILED
INFO    - 1/1 nrf52dk_nrf52832          tests/portability/cmsis_rtos_v1/portability.cmsis_rtos_v1 FAILED Failed (device 15.061s)
ERROR   - see: /home/maciej/zephyrproject2/zephyr/sanity-out/nrf52dk_nrf52832/tests/portability/cmsis_rtos_v1/portability.cmsis_rtos_v1/handler.log

INFO    - 0 of 1 tests passed (0.00%), 1 failed, 0 skipped with 0 warnings in 20.19 seconds
INFO    - In total 1 test cases were executed on 1 out of total 255 platforms (0.39%)
INFO    - 1 tests executed on platforms, 0 tests were only built.

Hardware distribution summary:

| Board            | ID   |   Counter |
|------------------|------|-----------|
| nrf52dk_nrf52832 |      |         1 |
```

**Environment:**
 - OS:  Linux, 
 - Toolchain Zephyr SDK
 - Commit SHA b5764f9902

**Additional context**
The test seems to pass on other platforms (https://testing.zephyrproject.org/daily_tests/v2.3.0-rc2/report/index.html). The value for `diff` supposed to be between `min 990000` and `max 1010000`
 From running it 3 times per platform I've got diff eq. to:
nrf52dk_nrf52832: 1032470, 1033355 and 1033843
nrf52840dk_nrf52840: 981140, 980743 and 981384 
nrf9160dk_nrf9160: 1127929, 1121948 and 1121459

PerMac, 2020-06-19:
The `diff` values per platform look precise (close to each other) but inaccurate (far from expected value). Could it be some calibration issue?

carlescufi, 2020-06-22:
@nordic-krch if you have a chance, could you take a quick look please?

PerMac, 2020-09-08:
The test continues to fail for nrf9160. The values for `diff` when failing are still close to the ones reported before (on my board, I am not sure if this value is the board- or platform- characteristic). As I mentioned in an earlier comment it looks like not a precision issue but an accuracy one. 
I would imagine that performing some clock calibration for a board as an initial step could solve these problems. Maybe this would also help with other clock related issues?

nordic-krch, 2020-09-09:
Here is the story: nrf9160 in practice will work only with LFXO (high precision) but when starting it initially runs from LFRC because LFXO has 200ms startup time and LFRC has 600uS. Because of that assumption (LFXO is a must) LFRC has no calibration and 30% tolerance. This test runs before switching to LFXO and because of that results are so poor.

problem would be fix by #27398 (require clock stabilization) but it won't get in for 2.4 since it is not a bugfix unless we push (@carlescufi ?). Workaround is to add busy wait there.

nordic-krch, 2020-09-09:
> The diff values per platform look precise (close to each other) but inaccurate

They will always be a bit inaccurate because busy wait runs from HF RC which has 8% accuracy. Additionally, even when LF XTAL is configured test may be executed before it is stabilized. Again this would be solved by #27398.

PR #26364 (2020-06-23): tests: portability: cmsis_rtos_v1: Increase clock tolerance
Increased accepted difference between system clock and
k_busy_wait clock. Some platforms (e.g. Nordic SoC) are using
different clocks for system clock (low frequency) and k_busy_wait
(high frequency). If those clocks are driven from low accuracy
source like RC oscillator then difference may be significant.

Fixes #26280.
PR #27398 (2020-09-10): nRF: Add system clock stability policy
Pull request adds clock stability policy configuration to the system clock (see #22758) and adds support to Nordic platform. Additionally, spin waiting for stable or available clock has been reworked for Nordic to allow going to idle while waiting for clock readiness (it can take up to 250 ms). Added test dedicated for Nordic platforms which validates correctness of behavior for each configuration against various clock sources.

Fixes #22758.
Fixes #26280.
closed 2020-09-10 by carlescufi