zephyrproject-rtos/zephyr #26291: samples: subsys : canbus: canopen:

wngur300, 2020-06-19:
**Describe the bug**
Execute canopen stack using mcp2515 driver at stm32l433 mcu and connect it with plc (XBC-DR232H) for testing.
If the CAN_MCP2515_MAX_FILTER value is greater than 8, Usage fault appears, and log error message appears if it is less than 8.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=stm32l433 ..
3. west flash
4. See error

*If the CAN_MCP2515_MAX_FILTER value is greater than 8,  The following error occurs:*
```
*** Booting Zephyr OS version 2.2.99  ***
[00:00:00.000,000] <inf> can_driver: Init of CAN_1 done
[00:00:00.005,000] <inf> fs_nvs: 8 Sectors of 2048 bytes
[00:00:00.005,000] <inf> fs_nvs: alloc wra: 1, 798
[00:00:00.005,000] <inf> fs_nvs: data wra: 1, b0
[00:00:00.006,000] <dbg> canopen_storage.canopen_settings_set: restored object dictionary EEPROM entries
[00:00:00.006,000] <dbg> canopen_driver.CO_CANmodule_init: rxSize = 13, txSize = 9
[00:00:00.007,000] <inf> app: CANopen stack initialized
[00:00:01.183,000] <inf> app: Resetting button press counter
[00:00:01.189,000] <err> os: ***** USAGE FAULT *****
[00:00:01.189,000] <err> os:   Illegal load of EXC_RETURN into PC
[00:00:01.189,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x0800d470  r2/a3:  0x200014ac
[00:00:01.189,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000008
[00:00:01.189,000] <err> os:  xpsr:  0x00000000
[00:00:01.189,000] <err> os: Faulting instruction address (r15/pc): 0x00000010
[00:00:01.189,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:01.189,000] <err> os: Current thread: 0x20000a40 (unknown)
[00:00:01.189,000] <err> os: Halting system
```

*If the CAN_MCP2515_MAX_FILTER value is less than 8, the following log error message can be viewed.*

```
*** Booting Zephyr OS version 2.2.99  ***
[00:00:00.000,000] <inf> can_driver: Init of CAN_1 done
[00:00:00.005,000] <inf> fs_nvs: 8 Sectors of 2048 bytes
[00:00:00.005,000] <inf> fs_nvs: alloc wra: 1, 788
[00:00:00.005,000] <inf> fs_nvs: data wra: 1, d0
[00:00:00.006,000] <dbg> canopen_storage.canopen_settings_set: restored object dictionary EEPROM entries
[00:00:00.006,000] <dbg> canopen_driver.CO_CANmodule_init: rxSize = 13, txSize = 9
[00:00:00.007,000] <inf> app: CANopen stack initialized
[00:00:01.895,000] <err> canopen_driver: failed to attach CAN rx isr, no free filter
[00:00:02.029,000] <err> canopen_driver: failed to attach CAN rx isr, no free filter
[00:00:02.042,000] <inf> app: Resetting button press counter
[00:00:02.048,000] <inf> app: Resetting button press counter
[00:00:02.054,000] <inf> app: Resetting button press counter
[00:00:02.060,000] <inf> app: Resetting button press counter
[00:00:02.067,000] <inf> app: Resetting button press counter
[00:00:02.073,000] <inf> app: Resetting button press counter
[00:00:02.079,000] <inf> app: Resetting button press counter
[00:00:02.085,000] <inf> app: Resetting button press counter
[00:00:02.091,000] <inf> app: Resetting button press counter
[00:00:02.098,000] <inf> app: Resetting button press counter
[00:00:02.104,000] <inf> app: Resetting button press counter
[00:00:02.110,000] <inf> app: Resetting button press counter
[00:00:02.116,000] <inf> app: Resetting button press counter
[00:00:02.122,000] <inf> app: Resetting button press counter
[00:00:02.128,000] <inf> app: Resetting button press counter
[00:00:02.134,000] <inf> app: Resetting button press counter
[00:00:02.141,000] <inf> app: Resetting button press counter
[00:00:02.147,000] <inf> app: Resetting button press counter
[00:00:02.340,000] <err> canopen_driver: failed to attach CAN rx isr, no free filter
[00:00:02.352,000] <inf> app: Resetting button press counter
[00:00:02.639,000] <err> canopen_driver: failed to attach CAN rx isr, no free filter
```

**Additional context**

Can_Stm32 driver was used to verify that the can open sample source was executed well by connecting plc (XBC-DR232H). I tried to run the canopen sample source using the mcp2515 driver, but the above problem occurred.

My device is the stm32l433RB, and I have added the following to the device tree file:
```
/ {
	chosen {
		zephyr,code-partition = &slot0_partition;
		zephyr,can-primary = &mcp2515_can;
	};

	leds {
		compatible = "gpio-leds";
		led0: led_0 {
			gpios = <&gpioc 9 GPIO_ACTIVE_LOW>;
			label = "User led0";
		};
		led1: led_1 {
			gpios = <&gpioa 8 GPIO_ACTIVE_LOW>;
			label = "User led1";
		};
	}; 

	gpio_keys {
		compatible = "gpio-keys";
		user_button: button_0 {
			label = "User SW0";
			gpios = <&gpioa 15 GPIO_ACTIVE_LOW>;
		};
	};

	aliases {
		green-led = &led0;
		red-led = &led1;
		sw0 = &user_button;
	};

};

&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 0x00004000>;
		};
		slot0_partition: partition@8000 {
			label = "image-0";
			reg = <0x00008000 0x00004000>;
		};
		slot1_partition: partition@c000 {
			label = "image-1";
			reg = <0x0000c000 0x00004000>;
		};
		scratch_partition: partition@10000 {
			label = "image-scratch";
			reg = <0x00010000 0x00004000>;
		};

		storage_partition: partition@14000 {
			label = "storage";
			reg = <0x00014000 0x0000c000>;
		};
	};
};

&spi1 {
	status = "okay";
	cs-gpios = <&gpioa 4 0>;

	mcp2515_can: mcp2515@0 {
		compatible = "microchip,mcp2515";
		spi-max-frequency = <1000000>;
		int-gpios = <&gpiob 0 GPIO_ACTIVE_LOW>;
		status = "okay";
		label = "mcp2515";
		reg = <0x0>;
		osc-freq = <16000000>;
		bus-speed = <125000>;
		sjw = <1>;
		prop-seg = <2>;
		phase-seg1 = <7>;
		phase-seg2 = <6>;
		#address-cells = <1>;
		#size-cells = <0>;
	};

};

&can1 {
	status = "okay";
	bus-speed = <125000>;
};
```

stm32l433.conf :
```
CONFIG_SPI_1=y
CONFIG_CAN_1=y
CONFIG_MPU_ALLOW_FLASH_WRITE=y
```

Next, we added mcp2515 MAX FILTER value to prj.config:
```
CONFIG_LOG=y
CONFIG_CANOPEN_LOG_LEVEL_DBG=y

CONFIG_SPI=y
CONFIG_CAN=y
CONFIG_CAN_INIT_PRIORITY=80
CONFIG_CAN_MCP2515=y
CONFIG_CAN_MAX_FILTER=13
CONFIG_CAN_MCP2515_MAX_FILTER=9

CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_NVS=y
CONFIG_SETTINGS=y
CONFIG_SETTINGS_NVS=y

CONFIG_CANOPEN=y
CONFIG_CANOPEN_SYNC_THREAD=y
CONFIG_CANOPEN_STORAGE=y
CONFIG_CANOPEN_LEDS=y

CONFIG_ARM_MPU=y
CONFIG_REBOOT=y
```

What should I do?
What do you think is the problem?

nixward, 2020-06-19:
@wngur300 you may need to increase the stack size for the mcp2515 driver thread or another thread in your system.

wngur300, 2020-06-19:
> @wngur300 you may need to increase the stack size for the mcp2515 driver thread or another thread in your system.

CONFIG_CAN_MCP2515_INT_THREAD_STACK_SIZE has already been changed to 1024, 2048, but the same error occurs.
Thank you for your reply.

nixward, 2020-06-19:
Can you figure out which thread `0x20000a40` is in your map file?

wngur300, 2020-06-19:
> Can you figure out which thread `0x20000a40` is in your map file?

I tried to find out what 0x20000a40 thread was.
I can't find it well.
Can you tell me how to check?

nixward, 2020-06-19:
Search for `20000a40` in your build folder in the file zephyr/zephyr.map

wngur300, 2020-06-22:
> Search for `20000a40` in your build folder in the file zephyr/zephyr.map

zephyr.map :
```
.bss.canopen_tx_queue
                0x0000000020000a20       0x10 subsys/canbus/canopen/libsubsys__canbus__canopen.a(CO_driver.c.obj)
                0x0000000020000a20                canopen_tx_queue
 .bss.canopen_tx_workq
                0x0000000020000a30       0x84 subsys/canbus/canopen/libsubsys__canbus__canopen.a(CO_driver.c.obj)
                0x0000000020000a30                canopen_tx_workq
 .bss._k_thread_obj_canopen_sync
                0x0000000020000ab4       0x74 subsys/canbus/canopen/libsubsys__canbus__canopen.a(canopen_sync.c.obj)
                0x0000000020000ab4                _k_thread_obj_canopen_sync
```

The same occurs when the value of CANOPEN_TX_WORKQUEUE_STACK_SIZE is increased.
Is there anything else I need to set up?

wngur300, 2020-06-22:
> Search for `20000a40` in your build folder in the file zephyr/zephyr.map

The CANOPEN_SYNC_THREAD_STACK_SIZE value is increased, so it works well. Thank you very much!!

nixward, 2020-06-22:
@wngur300 awesome! Great to hear!

closed 2020-06-30 by MaureenHelm