zephyrproject-rtos/zephyr #26295: Enable persistent storage (ext flash/SD card) as logger backend

koffes, 2020-06-19:
**Is your enhancement proposal related to a problem? Please describe.**
All log mechanisms today requires the user/developer to have either RTT/Serial/SWO connected to the board to see log output. Also, if you desire to store logs, this has to be done on the host side of the log link.

In most cases this is fine, but in some situations it would be highly desirable to place logs in on-board persistent storage:
A: A customer experiences an error, but did not have a backend connected. With this solution the log can be retrieved using a debugger from internal flash, USB from external flash or an SD card can be read over USB or put into a PC.

B: Logging from small portable devices. In many cases you have a portable device you want logging where it is cumbersome or impossible to have USB/UART/RTT connected.
E.g. logging battery level over a long period of time.

**Describe the solution you'd like**
LOG_BACKEND_PERSISTENT_STORAGE or
LOG_BACKEND_EXT_FLASH
LOG_BACKEND_SD_CARD
(LOG_BACKEND_INT_FLASH)

If the corresponding KConfig settings have not been made, you will get an error.
You will of course also have to set a file system, have a filename for the log and some kind of full storage handling.

**Alternatively**
As a first step, provide a LOG_BACKEND_USER or LOG_BACKEND_BLANK. When enabled, logs will e.g. go to UART and the USER backend at the same time. The user can then handle logs in the USER log callback as desired and the specifics of storing to flash or SD card can be left to the user.

koffes, 2020-06-19:
@nordic-krch

carlescufi, 2020-06-19:
This is actually a duplicate of #25907.

@koffes please add all the info there.

closed 2020-06-19 by carlescufi