zephyrproject-rtos/zephyr #26312: drivers: ieee802154: cc13xx_cc26xx: adopt hal/ti rf driverlib

cfriedt, 2020-06-19:
**Is your enhancement proposal related to a problem? Please describe.**
The ti hal rf driverlib and driver portability layer (dpl) likely contain most of what is necessary to support ieee 802.15.4 in Zephyr.

It might be advantageous to consider adopting the rf driverlib api so that there is api commonality with a common, vendor-supported hardware api.  

Another benefit is multi-band / multi-mac operation with ble.

**Describe the solution you'd like**
Refactor some common low-level / ti register apis with rf driverlib calls.

**Describe alternatives you've considered**
Just never doing that, and how it would be missing out on multi-mode operation.

**Additional context**

vanti, 2020-06-22:
@cfriedt Just to be clear, I think you meant the TI HAL's `RF driver `(the API defined in hal/ti/simplelink/source/ti/drivers/rf/RF.h) as opposed to `driverlib`, which is the package located under hal/ti/simplelink/source/ti/devices/cc13x2_cc26x2/driverlib - a library that gives low-level access to peripheral registers that Zephyr's ieee 802.15.4 driver is already relying on today.

cfriedt, 2020-06-26:
@vanti - yes, that's right, sorry.

cfriedt, 2020-09-03:
Going to get this moving again.

cfriedt, 2020-10-21:
Also blocks #21631

PR #26316 (2020-11-03): drivers: ieee802154: cc13xx_cc26xx: adopt hal/ti rf driverlib
This change reworks the cc13xx_cc26xx IEEE 802.15.4 driver to use the TI RF driver API that is available in modules/hal/ti.

There are a number of benefits to using TI's API including
 - a stable multi-OS vendor library and API
 - API compatibility with the rest of the SimpleLink SDK and SoC family
 - potential multi-protocol & multi-client radio operation (e.g. both 15.4 and BLE)
 - coexistence support with other chipsets via gpio
 - vetted TI RF driver resources, such as
   - the radio command queue
   - highly tuned / coupled RTC [... cut]
commit 8e2978d5 (2020-11-03): drivers: ieee802154: cc13xx_cc26xx: use ti rf driver api

This change reworks the cc13xx_cc26xx IEEE 802.15.4 driver to use
the TI RF driver API that is available in modules/hal/ti.

There are a number of benefits to using TI's API including
 - a stable multi-OS vendor library and API
 - API compatibility with the rest of the SimpleLink SDK and SoC family
 - potential multi-protocol & multi-client radio operation
   (e.g. both 15.4 and BLE)
 - coexistence support with other chipsets via gpio
 - vetted TI RF driver resources, such as
   - the radio command queue
   - highly tuned / coupled RTC  [... cut]
closed 2020-11-03 by carlescufi