zephyrproject-rtos/zephyr #26330: tcp: low bulk receive performance due to window handling

nzmichaelh, 2020-06-21:
If the remote device sends a burst of data that fills the TCP receive window, then reception halts until the remote sends a keep-alive.  This reduces the peak receive throughput by > 10x.

The sequence is:

- Zephyr advertises a receive window of 1280 bytes
- Remote sends one packet of 640 bytes
- Remote sends another packet of 640 bytes
- Zephyr acknowledges with a receive window of 0 bytes
- The Zephyr app reads the data
- Later, the remote sends a TCP keep-alive
- Zephyr acknowledges with a receive window of 1280 bytes

The remote waits ~100 of ms before sending the keep-alive, which limits the throughput to ~12800 B/s.

Ideally, if the window was full, then Zephyr would send a packet with the updated window once the app starts reading data.

This might be related to #23302

Context: I'm writing a VNC client using Zephyr.  The VNC server sends large bitmaps down to Zephyr over a IPv4 over a USB ECM connection.  I'm testing on the Arduino Zero.

jukkar, 2020-06-22:
@nzmichaelh would you be able to try the new TCP implementation as it has some improvements in this area, just set `CONFIG_NET_TCP2=y` in your config file.

nzmichaelh, 2020-06-23:
Hi @jukkar I wrote a simple TCP discard server and tried it with TCP1 and TCP2.  TCP1 shows the window-is-full issue.  TCP2 has a variety of issues:

- The stack does unaligned access, which causes a fault on the Cortex-M0+
- The stack doesn't seem to implement receive windows.  The board always reports a window of 1280 bytes, so Linux keeps sending data, which eventually causes the board to run out of RX buffers
- The stack doesn't cleanly close down the connection.  When Linux sends a FIN, the board seems to drop the packet.

I'll send a PR for the unaligned access faults.

jukkar, 2020-06-23:
@nzmichaelh Thanks for the report.

> The stack doesn't cleanly close down the connection.  When Linux sends a FIN, the board seems to drop the packet.

@ozhuraki can you check this one.

armandciejak, 2020-07-09:
Hello,

We are having a system, which uses the i.MXRT1062, in production and this problem makes or firmware upgrade process extremely slow (15 minutes instead of 30 seconds).

Is it possible to raise the priority of this issue?

Is there any workaround we can apply until this is properly fixed?

Thanks

nzmichaelh, 2020-07-09:
Hi @armandciejak I saw some improvement by changing `NET_TCP_BUF_MAX_LEN` in tcp_internal.h and change to `(4*1280)`.  That way you'll receive 4x the data before the window is full and, if lucky, the host will hardly ever fill the whole window.

I also found that doing many small reads kept the window fresher.

Both of these are hacks and only statistically improve the throughput, but are worth a try.

armandciejak, 2020-07-15:
Hi @nzmichaelh, I've tried increasing `NET_TCP_BUF_MAX_LEN` to 4096 and that helps a bit but not with all OS (Windows 10 being the worst).

Here is what I observed:
1. Server (Zephyr application) advertise a window size of 4096
1. Client (Chrome browser on Win10) sends 7 segment of 536 bytes
1. Server advertise a window size of 344 bytes (in the ack packet for the last segment)
1. Client stops sending for 5 seconds
1. After the 5 seconds the client finally sends a segment of 344 bytes
1. Server advertise a window size of 3752 bytes
1. Goto 2.

You can see one iteration of the above loop in the wireshark screenshot below (time is "second since last displayed packet").

[image]

It takes 5 seconds to send 3752 bytes, which is extremely slow (around 750 bytes/sec).

We looked at the problem for several days with Wireshark and testing the behavior of several different OS. Windows 10 is the worst but other OS shows the same problem with either lower "pause" time or less often but uploading "big" files it still barely usable.

armandciejak, 2020-07-15:
@jukkar @ozhuraki do you know if it is possible to trigger sending a window update packet from the server in order to unblock the client?

*Note: I hacked the TCP stack to send a window update packet when the application (web server) consumes data and that effectively unblocks the client. This hack was done to observe the behavior of the client, everything was hard-coded therefore it is not worth sharing it.*

nzmichaelh, 2020-07-15:
@armandciejak wow, that's pretty unusable.  Linux sends a a probe after ~0.5s so it's a bit better.

carlescufi, 2020-09-08:
@nzmichaelh could you retest now that TCP2 is the default in Zephyr please? Please close the issue if this is resolved.

jukkar, 2020-09-29:
@nzmichaelh there has been improvements with window handling, can you check if it helps your use cases?

closed 2020-12-13 by github-actions[bot]