zephyrproject-rtos/zephyr #26333: Bluetooth: Split LL: Cannot store Bluetooth keys

MarekPieta, 2020-06-22:
**Describe the bug**
I used Zephyr's Split LL on nrf_desktop dongle (Bluetooth central). I noticed that in some cases the dongle failed to store Bluetooth keys after a new bonding when it already had other connected peripherals.  The problem is related to storing data in non-volatile memory on Zephyr Bluetooth central that has multiple connections.

**To Reproduce**
Issue was replicated using nrf_desktop application with increased connection count on central. Please use FW from this branch:
https://github.com/MarekPieta/fw-nrfconnect-nrf/tree/splitLL_settings_issue

1. Flash the dongle (you can use nrf52840dongle_nrf52840 or nrf52840dk_nrf52840; on the branch I switched to using Split LL in the ZDebug_dongle configuration for nrf52840dk_nrf52840)
`cmake .. -DBOARD=nrf52840dongle_nrf52840 -DCMAKE_BUILD_TYPE=ZDebugSplitLL -GNinja
`
`cmake .. -DBOARD=nrf52840dk_nrf52840 -DCMAKE_BUILD_TYPE=ZDebug_dongle -GNinja
`
2. Flash the peripherals. You can connect up to 4 peripherals to the dongle (the dongle with FW from the linked branch can connect with multiple peripherals of the same type – for example with multiple mice). The FLASH issue on the dongle (Bluetooth central) replicates more often when more peripherals are connected.
Use the following cmake command to build FW for nrf52840dk_nrf52840 (the ZDebug configuration for nrf52840dk_nrf52840 already has the `CONFIG_BT_SMP_ALLOW_UNAUTH_OVERWRITE=y`, make sure to set this option if you use other board/configuration as peripheral):
`cmake .. -DBOARD=nrf52840dk_nrf52840 -DCMAKE_BUILD_TYPE=ZDebug -GNinja
`
3. Turn on the peripherals and the dongle. The dongle will automatically scan for the peripherals and connect with them. You may observe the FLASH issue during pairing (the dongle stores data related to the pairing in non-volatile memory).
4. You can short press the button on dongle (or button 1 on DK) to trigger scanning in case it was interrupted by a timeout or using one of the connected peripherals.
5. You can press and hold the button to erase all the bonds. When the LED starts blinking you can release the button and then double click it to confirm erase. Then the bonds are erased and the dongle will overwrite the old bonding (`CONFIG_BT_SMP_ALLOW_UNAUTH_OVERWRITE=y` on peripheral side) when connecting with the peripheral. This triggers FLASH operations and gives opportunity to replicate the issue.

In nrf_desktop application logs are sent over the RTT (debug configurations).

**Expected behavior**
Dongle should properly complete FLASH operations with multiple Bluetooth connections.

**Impact**
Zephyr based Bluetooth central using Split LL, with multiple Bluetooth connections does not support FLASH operations.

**Screenshots or console output**
I verified that there is a timeout: `work_in_time_slice` in `drivers/flash/soc_flash_nrf.c` returns `-ETIMEDOUT`. I increased the `FLASH_TIMEOUT_MS` (multiplied it by 4 or 10), but this did not fix the FLASH issues. It looks like Bluetooth operations still somehow do not allow to perform FLASH operations.

**Environment**
 - Toolchain: Zephyr SDK: 0.11.1
- Commit hash: Please use the code from branch

**Additional context**
The problem did not occur for nrfxlib Link Layer. It's specific to Zephyr's Split LL.

carlescufi, 2020-06-22:
@cvinayak and @nvlsianpu this looks like an issue in either the flash driver or the ticker.

cvinayak, 2020-07-08:
Reproduced like 1 in 5 bond erase when having 4 peripheral connect and bond.

[image]

This (again) appears to be new ticker implementation issue, two different connection event when overlapping with flash drive timespace, they dont round-robin and connection events appear to be given precedence starving the flash driver.

I am reviewing the ticker implementation for a solution...

mtpr-ot, 2020-07-08:
@cvinayak, did you use the suggested reproduction scenario, or did you find an easier way to reproduce this? I assume the flash ticker does not use slot reservation, and is periodic?

cvinayak, 2020-07-08:
@mtpr-ot On analyzing the ticker implementation in details, I dont find anything wrong. Instead, reviewing the nrf flash driver revealed that the driver was using ticker id 0 which overlaps with split controller's pipeline preempt ticker id. This overlap can definitely cause flash driver to stall.

I will test more the flash driver fix using an explicit new ticker id and send a PR to resolve this issue.

@mtpr-ot regret the misunderstanding caused. flash driver does use slot reservation, but also uses a secondary ticker to abort radio use. This secondary ticker reused the pipeline preempt ticker id 0 (this was ok with in the legacy controller).

PR #26754 (2020-07-09): drivers: flash: nrf: Fix flash operation timeout
Fix flash operation timeout due to incorrect use of
secondary ticker to abort any radio in use. Ticker id 0
is reserved for split controller's pipeline preempt timeout.
Using the same ticker id caused the secondary ticker to
not be started if controller is using the same ticker id
for pipeline preempt timeout.

Fixes #26333.
commit eb2a6498 (2020-07-09): drivers: flash: nrf: Fix flash operation timeout

Fix flash operation timeout due to incorrect use of
secondary ticker to abort any radio in use. Ticker id 0
is reserved for split controller's pipeline preempt timeout.
Using the same ticker id caused the secondary ticker to
not be started if controller is using the same ticker id
for pipeline preempt timeout.

Fixes #26333.
closed 2020-07-09 by nashif