zephyrproject-rtos/zephyr #26337: BT scan: filter duplicates yields duplicates

spotperg, 2020-06-22:
**Describe the bug**
Scanning for BT devs with filter for duplicates enabled no longer gives the expected behavior. I have found that setting CONFIG_BT_CTLR_DUP_FILTER_LEN (default 16) to some large number (128, larger than the amount of (devs * AD types) around you) gives the expected behavior. The list of found devices is treated as circular, so given that the number of devices * AD types around you is larger than the list length, the list of found devices will be constantly re-written. 

I have no idea what has changed since 2.2, but this behavior came with 2.3.

**To Reproduce**
Compile and run tests/bluetooth/shell (I use west)
west build zephyr/tests/bluetooth/shell -b nrf52833dk_nrf52833
west flash
Open terminal and connect to dev kit; once connected enter commands:
bt init
bt scan on nodups
See the scan go on forever...

Recompile with "CONFIG_BT_CTLR_DUP_FILTER_LEN=128" (e.g.) in your prj.conf, and the scan does not yield duplicates.

**Expected behavior**
A scan that does not yield duplicates. 

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain GNU arm embedded
 - Commit 9bab3a84b6427662a1d3df45db365d23f2eeaa3a (as part of NRF SDK)

joerchan, 2020-06-22:
The default list size has been 16 for a while (2017 according to git). I see the same behavior on earlier versions. Maybe there is just more devices around you now than before? If the list is exceeded then duplicates will show. That is by design and should not be treated as a bug.

spotperg, 2020-06-22:
I feel like something has changed, but I am not able to pinpoint what. Maybe it's just an odd corona period thing, with fewer devices around? Idk. 

I am happy with this explanation and the possibility to set a manual size for the filter list.

closed 2020-06-22 by spotperg