zephyrproject-rtos/zephyr #26343: Gatt Bearer Issue

johnsmithecon, 2020-06-22:
**Describe the bug**
Hi all, I think I might have found a glitch in the mesh SDK (i am noob so maybe I'm making the mistake). The way I reproduce it is I use the nrfmeshprovision SDK for iOS to open a gatt bearer with a node. I successfully open the gatt bearer. Then I rerun the app and open gatt bearer again using nrfmeshprovision SDK with that same node. I am able to connect to the node using the bearer, then discover services, but I am unable to open the bearer. This is because the node crashed! The node actually crashes with the following error:
```
[00:00:17.315,887] [1;33m<wrn> bt_mesh_proxy: Client wrote 0x0000 instead enabling notify[0m
[00:00:27.785,980] [1;31m<err> os: ***** MPU FAULT *****[0m
[00:00:27.785,980] [1;31m<err> os:   Instruction Access Violation[0m
[00:00:27.785,980] [1;31m<err> os: r0/a1:  0x0002b37c  r1/a2:  0x00000000  r2/a3:  0x20000d20[0m
[00:00:27.786,010] [1;31m<err> os: r3/a4:  0x20004c48 r12/ip:  0x00000064 r14/lr:  0x000192c3[0m
[00:00:27.786,010] [1;31m<err> os:  xpsr:  0x20000000[0m
[00:00:27.786,010] [1;31m<err> os: Faulting instruction address (r15/pc): 0x20004c48[0m
[00:00:27.786,010] [1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0[0m
[00:00:27.786,010] [1;31m<err> os: Current thread: 0x2000128c (unknown)[0m
[00:00:29.188,842] [1;31m<err> os: Halting system[0m
```

**Environment (please complete the following information):**
MacOS, nRF52 DK

**Additional context**
Add any other context about the problem here.

trond-snekvik, 2020-06-22:
Hey, are you able to get the location the link register is pointing to?
You can for instance try addr2line:

```
arm-none-eabi-addr2line -e build/zephyr/zephyr.elf -a 0x192c3
```

Or you can connect gdb (for instance with `west attach`) and call list:

```
list *0x192c3
```

Similar output for the addresses is r0 and current thread could also be of interest.

If you're able to reproduce this easily, you could try increasing stack sizes as well, to see if it's an overflow.

johnsmithecon, 2020-06-22:
0x192c3 is in isr_done (/.../zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv.c:472).
[image]

closed 2020-06-23 by carlescufi

LingaoM, 2020-06-26:
I can't see the relationship between this patch and issue.

trond-snekvik, 2020-06-26:
@LingaoM The GATT callbacks were cleared in the state_reset function that got called on disconnect. This caused the device to crash on the next connection, as it called the callback unchecked. The code assumed that the callback could not be NULL as long as provisioning was active. This assumption is now true

PR #26379 (2020-06-23): Bluetooth: Mesh: Keep PB-GATT callback through disconnect
Removes the memset of the prov_bearer_cb in PB-GATT during resets. This
allows the provisioning link to disconnect and reconnect again without
having to call pb_gatt_open.

Fixes #26343.
commit 5ea787c7 (2020-06-23): Bluetooth: Mesh: Keep PB-GATT callback through disconnect

Removes the memset of the prov_bearer_cb in PB-GATT during resets. This
allows the provisioning link to disconnect and reconnect again without
having to call pb_gatt_open.

Fixes #26343.