zephyrproject-rtos/zephyr #26353: [Coverity CID :211039] Out-of-bounds access in drivers/gpio/gpio_nrfx.c

zephyrbot, 2020-06-23:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/8e2c4a475dc375da6691175dd1da87525053ed76/drivers/gpio/gpio_nrfx.c#L105

Category: Memory - corruptions
Function: `gpiote_pin_int_cfg`
Component: Drivers
CID: [211039](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=211039)

Details:

```
99     {
100         struct gpio_nrfx_data *data = get_port_data(port);
101         const struct gpio_nrfx_cfg *cfg = get_port_cfg(port);
102         uint32_t abs_pin = NRF_GPIO_PIN_MAP(cfg->port_num, pin);
103         int res = 0;
104    
>>>     CID 211039:  Memory - corruptions  (ARRAY_VS_SINGLETON)
>>>     Passing "&gpiote_alloc_mask" to function "gpiote_pin_cleanup" which uses it as an array. This might corrupt or misinterpret adjacent memory locations.
105         gpiote_pin_cleanup(&gpiote_alloc_mask, abs_pin);
106         nrf_gpio_cfg_sense_set(abs_pin, NRF_GPIO_PIN_NOSENSE);
107    
108         /* Pins trigger interrupts only if pin has been configured to do so */
109         if (data->pin_int_en & BIT(pin)) {
110             if (data->trig_edge & BIT(pin)) {
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

pabigot, 2020-06-29:
I propose marking this issue as False Positive/minor/ignore in Coverity with the following analysis.  @anangl do you concur?

This is an atomic_t value supporting 32 bits, being used in a loop where the iteration maximum `GPIOTE_CH_NUM` is at most 8.  The implementation will not access outside the declared single atomic_t value.

anangl, 2020-06-30:
@pabigot I agree with the above analysis. But perhaps it would be cleaner to replace the following line in `gpiote_pin_cleanup()`
https://github.com/zephyrproject-rtos/zephyr/blob/8e2c4a475dc375da6691175dd1da87525053ed76/drivers/gpio/gpio_nrfx.c#L81
with
```
(void)atomic_and(mask, ~BIT(i));
```
and this way avoid that array confusion?
This would be also consistent with how bits are set for allocated channels (`atomic_or()`). What you think?

pabigot, 2020-06-30:
Consistency is better.  Will you do a PR for that or do you want me to?

anangl, 2020-06-30:
I will do it.

PR #26535 (2020-07-06): drivers: gpio_nrfx: Replace atomic_set_bit_to() with atomic_and()
A call to atomic_set_bit_to() was used for clearing bits in an atomic_t
variable that stores information about allocated GPIOTE channels.
This caused an issue to be reported by Coverity, as the function treats
its first parameter as an array and it was provided with a pointer to
a singleton.
This commit replaces that call with atomic_and(), to prevent the issue
from being reported and for consistency with the way bits are set for
allocated GPIOTE channels (what is done with a call to atomic_or()).

Fixes #26353.
closed 2020-07-06 by carlescufi