zephyrproject-rtos/zephyr #26355: [Coverity CID :211045] Macro compares unsigned to 0 in kernel/timeout.c

zephyrbot, 2020-06-23:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/8e2c4a475dc375da6691175dd1da87525053ed76/kernel/timeout.c#L112

Category: Integer handling issues
Function: `z_add_timeout`
Component: Kernel
CID: [211045](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=211045)

Details:

```
106    
107         LOCKED(&timeout_lock) {
108             struct _timeout *t;
109    
110             to->dticks = ticks + elapsed();
111             for (t = first(); t != NULL; t = next(t)) {
>>>     CID 211045:  Integer handling issues  (NO_EFFECT)
>>>     This greater-than-or-equal-to-zero comparison of an unsigned value is always true. "t->dticks >= 0ULL".
112                 __ASSERT(t->dticks >= 0, "");
113    
114                 if (t->dticks > to->dticks) {
115                     t->dticks -= to->dticks;
116                     sys_dlist_insert(&t->node, &to->node);
117                     break;
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

dleach02, 2020-08-14:
@andyross, This assert doesn't make sense? Do you know what the intent was? Was dticks once a signed value?

andrewboie, 2020-08-14:
is this an artifact when time values used to be signed?

dleach02, 2020-08-15:
dticks used to be unsigned so likely a left over. I'll add a fix to my PR I'm putting together to clean up these coverity issues

PR #27591 (2020-08-16): Address multiple Coverity issues
Multiple coverity issues are addressed in this PR.

Fixes #26355
Fixes #25792
Fixes #26998
Fixes #26997
Fixes #26359
Fixes #26358
commit 1ab90044 (2020-08-16): kernel/timeout: Fix coverity warning CID 211045

The dticks field was changed from a signed to an unsigned
value so now the assert test to ensure it isn't negative
is no longer needed.

fixes #26355
closed 2020-08-16 by nashif