zephyrproject-rtos/zephyr #26362: arc gdb failed to load core dump file

clear-zhang, 2020-06-23:
**Describe the bug**
Wen we try to use gdb for arc cpu to load core dump file. Below error occurs.
"/home/clzhang/cvf_ww14/pse/tmp/coredump.elf": no core file handler recognizes format

clzhang@clzhang-jlf-simics:~/cvf_ww14/pse$ arc-zephyr-elf-gdb ./pub/cvf/arc/fpga_em7d_fw/zephyr.elf tmp/coredump.elf
GNU gdb (crosstool-NG 1.24.0-rc2-dirty) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=arc-zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./pub/cvf/arc/fpga_em7d_fw/zephyr.elf...done.
"/home/clzhang/cvf_ww14/pse/tmp/coredump.elf": no core file handler recognizes format
(gdb)

**To Reproduce**
Steps to reproduce the behavior:
1. In application, do a null pointer operation.
2. in fatal.c, dump a core dump file
3. run arc-zephyr-elf-gdb command for arc in SDk: arc-zephyr-elf-gdb ./pub/cvf/arc/fpga_em7d_fw/zephyr.elf tmp/coredump.elf 
4. See error

**Expected behavior**
core dump file is recognized and loaded.

**Impact**
It's high since it is blocking core dump feature development. 

**Screenshots or console output**
clzhang@clzhang-jlf-simics:~/cvf_ww14/pse$ arc-zephyr-elf-gdb ./pub/cvf/arc/fpga_em7d_fw/zephyr.elf tmp/coredump.elf
GNU gdb (crosstool-NG 1.24.0-rc2-dirty) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=arc-zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./pub/cvf/arc/fpga_em7d_fw/zephyr.elf...done.
"/home/clzhang/cvf_ww14/pse/tmp/coredump.elf": no core file handler recognizes format
(gdb)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (e.g Zephyr SDK, ...)
 - clzhang@clzhang-jlf-simics:~/cvf_ww14/pse$ /opt/zephyr-sdk/arc-zephyr-elf/bin/arc-zephyr-elf-gdb --version
GNU gdb (crosstool-NG 1.24.0-rc2-dirty) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

**Additional context**
Add any other context about the problem here.

clear-zhang, 2020-06-23:
[](url)

clear-zhang, 2020-06-23:
[coredump.txt](https://github.com/zephyrproject-rtos/zephyr/files/4816742/coredump.txt)
this is a core dump file created by me. please change it as *.elf

wentongwu, 2020-06-23:
@galak

clear-zhang, 2020-07-13:
I found arc-linux-gdb has no this problem while arc-elf-gdb has. Seems it's caused by the different target option when gbd is built. Hope this can provide information for you

ruuddw, 2020-07-13:
Thanks @clear-zhang, this helps (and would be a workaround for now). The tools team told me earlier this was a feature they added specifically for Linux core dumps, so maybe indeed a matter of setting the right flags when building - will check with them.

shahab-vahedi, 2020-07-13:
@clear-zhang thanks for all the detailed info. Indeed, `arc-linux-gdb` is able to parse those coredumps, because that was the foreseeable scenario:

1. A userspace program running inside GNU/Linux OS coredumps.
2. `arc-linux-gdb` must be able to parse it.

For now, please continue using `arc-linux-gdb`. If you run into any trouble, please do let me know.

Track the implementation of the feature is [here](https://github.com/foss-for-synopsys-dwc-arc-processors/binutils-gdb/issues/36).

galak, 2020-07-22:
Need someone from the arc side to comment on if there are changes/patches that ARC is support for GDB for this.

shahab-vahedi, 2020-07-22:
@galak I don't understand. What else do you need that the previous comment doesn't make clear?

shahab-vahedi, 2020-07-28:
> **To Reproduce**
> Steps to reproduce the behavior:
> 
>     1. In application, do a null pointer operation.
> 
>     2. in fatal.c, dump a core dump file
> 
>     3. run arc-zephyr-elf-gdb command for arc in SDk: arc-zephyr-elf-gdb ./pub/cvf/axx/fpga_exxx_fw/zephyr.elf tmp/coredump.elf
> 
>     4. See error

I cannot get the `coredump` generated. What do you exactly mean by _step 2_? I looked a little bit into [ARC's fatal.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/arch/arc/core/fatal.c) and thought maybe I should have `ARC_EXCEPTION_DEBUG` enabled, but that didn't change anything.

I know you have [provided the coredump](https://github.com/zephyrproject-rtos/zephyr/issues/26362#issuecomment-647871648), but I rather use one that comes from the elf file I have. For now, I am actually using the provided [coredump](https://github.com/zephyrproject-rtos/zephyr/files/4816742/coredump.txt) against an arbitrary `zephyr.elf`.

To be complete, I used this [Zephyr](https://github.com/foss-for-synopsys-dwc-arc-processors/zephyr/commit/cbb252adef08ba482b14ef2ab562baf879956c93).

shahab-vahedi, 2020-07-28:
I looked into other targets such as ARM. From the code, it looks like gdb should behave the same for them: `*-elf*-gdb` should not be able to parse the coredump while the `*-linux-gdb` should.

Could someone tell me how the *elf* gdb behave for other targets?

galak, 2020-07-28:
> Thanks @clear-zhang, this helps (and would be a workaround for now). The tools team told me earlier this was a feature they added specifically for Linux core dumps, so maybe indeed a matter of setting the right flags when building - will check with them.

@ruuddw let me know and I can update the build configuration in the Zephyr SDK for ARC.

abrodkin, 2020-07-30:
@clear-zhang @galak let's first make it a bit more clear what is the problem here. For instance I don't quite understand how that core dump is supposed to be created? Which target is used for that, how do we interact with that target. In case of the Linux-based OS core dump is being created by the kernel itself and gets put on the filesystem. How that could be done in case of Zephyr is not clear to me.

galak, 2020-08-04:
Lower priority as @abrodkin mentioned I don't see how a core dump could even be generated w/Zephyr.

ruuddw, 2020-08-14:
@clear-zhang Can you please explain how you generated the coredump? Did you dump it from gdb (generate-core-file) or so?

abrodkin, 2020-09-07:
I'm closing it now since we don't really understand what's the problem here and for more than a month there were no updates with clarification. Feel free to re-open it once more info is avaialble.

closed 2020-09-07 by abrodkin