zephyrproject-rtos/zephyr #26363: samples: subsys: canbus: canopen: objdict: CO_OD.h is not normally made.

wngur300, 2020-06-23:
**Describe the bug**
The CO_OD.h file is not created normally in the Object Dictionary Editor v0.8-0-gb60f4b version. (CO_OD.h: #define CO_NO_TS     1)

**To Reproduce**
Steps to reproduce the behavior:
1. Run the Object Dictionary Editor v0.8-0-gb60f4b program.
2. Open the eds file provided by the canopen sample.
3. Create a CO_OD.c , CO_OD.h, xml file without any further changes, and save the eds file.
4. Place the generated CO_OD.c , CO_OD.h, and xml files in the canopen/objdict folder.
5. cmake -DBOARD=stm32l433 .. (canopen/build)
6. west flash

**Expected behavior**
```
(Unbuilt header files created by the editor program)
CO_OD.h: #define CO_NO_TS     1

(Header file for normal build)
CO_OD.h: #define CO_NO_TIME     1
```

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
- OS: Linux(Ubuntu 18.04)
- Toolchain: Zephyr SDK 0.11.1

**Additional context**
I wanted to create CO_OD.c, CO_OD.h, xml files directly by opening the eds file provided by the zipper canopen sample with object logic editor.
Download the editor program from (https://github.com/robincornelius/libedssharp/releases.)
We opened the editor program to create CO_OD.c, CO_OD.h, and xml files without any additional changes.
However, an error occurred during the build because `CO_NO_TIME` was not defined in the CO_OD.h file.
Looking in the header file,I suspected that the
```
#define CO_NO_TS      1
```
was created incorrectly due to the difference in the version of the editor.

I wonder if there is anything else that needs to be set up in the editor program in order to be normally generated from CO_OD.h to #define CO_NO_TIME    1 or if it is due to version differences.

henrikbrixandersen, 2020-06-23:
@wngur300 You need to use a version of libedssharp, that matches the version of CANopenNode used. This is not specific to Zephyr.

henrikbrixandersen, 2020-06-23:
To clarify, this is v0.8 of libedssharp (the latest released version). See https://github.com/robincornelius/libedssharp/releases/tag/v0.8

wngur300, 2020-06-23:
> @wngur300 You need to use a version of libedssharp, that matches the version of CANopenNode used. This is not specific to Zephyr.

v0.8-7-g2e53bbc is not found at https://github.com/robincornelius/libedssharp/releases and appears to be downloadable until version v0.8-0-gb60f4. Can you tell me where I can download the version of v0.8-7-g2e53bbc? Can't I get it for free?

henrikbrixandersen, 2020-06-23:
> v0.8-7-g2e53bbc is not found at https://github.com/robincornelius/libedssharp/releases and appears to be downloadable until version v0.8-0-gb60f4. Can you tell me where I can download the version of v0.8-7-g2e53bbc? Can't I get it for free?

To resolve the above issue, you will need at least https://github.com/robincornelius/libedssharp/commit/07ac8191e64067547f6c21774b5089ad03a465c2 (or just manually edit the generated file to use `CO_NO_TIME`.

wngur300, 2020-06-24:
> > v0.8-7-g2e53bbc is not found at https://github.com/robincornelius/libedssharp/releases and appears to be downloadable until version v0.8-0-gb60f4. Can you tell me where I can download the version of v0.8-7-g2e53bbc? Can't I get it for free?
> 
> To resolve the above issue, you will need at least [robincornelius/libedssharp@07ac819](https://github.com/robincornelius/libedssharp/commit/07ac8191e64067547f6c21774b5089ad03a465c2) (or just manually edit the generated file to use `CO_NO_TIME`.

Thank you!

carlescufi, 2020-06-24:
Closing as resolved.

closed 2020-06-24 by carlescufi

Qbicz, 2021-03-08:
For reference (and Linux users), the latest auto-build that has this problem fixed can be downloaded from: https://github.com/robincornelius/libedssharp/raw/gh-pages/build/OpenEDSEditor-latest.zip

Qbicz, 2021-03-08:
Actually the latest version of OpenEDSEditor is also not compatible, so I opened https://github.com/zephyrproject-rtos/zephyr/issues/33149 to clarify and find a solution.