zephyrproject-rtos/zephyr #26372: qspi driver does not work if multithreading is disabled

pdunaj, 2020-06-23:
**Describe the bug**
Build `zephyr/samples/drivers/spi_flash`. Check that it works. Disable multithreading and observe crash with MPU fault.

I have modified the sample so it only performs the read operation (I found the problem while working with mcuboot, when it tries to read image header). I see that all addresses are set but at some point driver waits for completion (k_sem_take). System crashes at some point after this is called or when interrupt handler tries to restore main.

**To Reproduce**
Steps to reproduce the behavior:
In description.

**Expected behavior**
The driver should work. If it does not it should depend on specific system options that are required.

**Impact**
I need this driver in mcuboot. So far the bootloader works without multithreading. If I would have to enabled the multithreading the size of the bootloader will increase so the bug can be considered critical.

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
master

**Additional context**
N/A

pdunaj, 2020-06-23:
@anangl

carlescufi, 2020-06-23:
This driver was never designed to work without threading, so it will require changes if we are to support non-multithreading.

pdunaj, 2020-06-23:
Hi @carlescufi , that explains the behavior. It would be great if it had a `depend on` to avoid confusion. For now I simply enabled multithreading.

pabigot, 2020-06-23:
It might not be too difficult to rework this to use `sys_notify` and select the spin-wait or callback notification mechanisms if multithreading isn't available.

pabigot, 2020-06-29:
It should be trivial to rework this to busy-wait, but it appears that if `CONFIG_MULTITHREADING=n` then something prevents `nrfx_qspi_irq_handler()` from being invoked when the `QSPI` peripheral `EVENT_READY` is triggered.

pabigot, 2020-09-02:
Given the decision on  #27415 to deprecate `CONFIG_MULTITHREADING=n` there doesn't seem any reason to keep this open.

pabigot, 2020-09-28:
Reopening as there's a chance #28679 will enable this to be fixed.

PR #28740 (2020-11-28): drivers: flash: nrf_qspi_nor: remove multithreading dependency
Replace semaphores with an atomic ready flag when used without multithreading enabled.

DNM until dependent PR is merged:
- [x] #28679 

Fixes #26372

Tested manually on samples/drivers/spi_flash and samples/subsys/fs/littlefs.
closed 2020-11-28 by nashif