zephyrproject-rtos/zephyr #26377: Problems getting I2C to work on NXP i.MX RT1020 EVK

puchm, 2020-06-23:
Hi,

I have already posted this question to the NXP forum but I figured it would make sense to post it here as well.
I am trying to set up the i.MX RT1020 so that it can send I²C signals to an Arduino. Here's some information on what I am doing so far:

Part number: MIMXRT1021DAG5A
OS: Zephyr 2.3.0-rc1
Pins: I am using the D14 and D15 pins in the J19 area for SDA (D14) and SCL (D15)

This is my code:
```
#include <zephyr.h>
#include <sys/printk.h>
#include <device.h>
#include <drivers/i2c.h>

#define DESTINATION_ADDR 0x34
#define HIGH_VALUE 0x01

void main(void) {
   struct device *dev;
   int ret;

   uint8_t data = HIGH_VALUE;

   dev = device_get_binding("I2C_1");
   if(!dev) {
      printk("I2C: Device not found\n");
      return;
   } else {
      printk("I2C: Device found\n");
   }
   while(true) {
      printk("Trying...");
      ret = i2c_write(dev, &data, sizeof(data), DESTINATION_ADDR);
      printk("done");
      k_msleep(1000);
   }
}
```

I am compiling using: west build -p auto -b mimxrt1020_evk
The code compiles without any errors.
When running, it logs "Device found" to my serial console and then repeatedly logs "Trying..." and "done", so the while loop is also getting run.

I am however not getting any I²C signals on my Arduino. I am quite sure that the problem is on the side of the RT1020 since the Arduino would log all I²C inputs to another serial console but it doesn't show anything. 

What am I doing wrong? My feeling is that I am either using the wrong ports or there is something wrong with my code.

I hope that I provided enough information for you to understand my issue. Thank you for your help in advance.

-Moritz

MaureenHelm, 2020-06-25:
I2C_4 is connected to Arduino D14 and D15; I2C_1 is connected to Arduino A4 and A5. You need to switch the instance string in your code, or rewire your boards to use A4 and A5.

pabigot, 2020-08-19:
I'm going to assume that @MaureenHelm's answer resolved this.  If there are further problems please open a new issue.

closed 2020-08-19 by pabigot