zephyrproject-rtos/zephyr #26383: OpenThread NCP radio-only

marvinroger, 2020-06-23:
**Is your feature request related to a problem? Please describe.**
I am trying to build an OpenThread RCP (radio-only RCP) on nRF52840, and I saw that there's a `CONFIG_OPENTHREAD_NCP_RADIO` config, which is exactly what I'm looking for. But the only reference of `OPENTHREAD_NCP_RADIO` that I find in the Zephyr code is the `Kconfig`.

**Describe the solution you'd like**
Given the fact that the config does not have any effect (unless I am missing something?), I am not sure if this is a bug or a feature request.

Thanks for your work!

rlubos, 2020-06-24:
The fact that `OPENTHREAD_NCP_RADIO` is not referenced anywhere is clearly a bug.

Another thing is, that OpenThread CMake options we used for NCP configuration in Zephyr were removed from the upstream OpenThread, and this change was merged with one of the recent OpenThread updates (and went unnoticed so far). This needs a closer look.

CC @markus-becker-tridonic-com

markus-becker-tridonic-com, 2020-06-25:
@rlubos I believe upstream has dropped NCP in https://github.com/openthread/openthread/commit/9a3c2dc3b373da5f3383720acc5cf8ece8368a31#diff-21cd1d48e2fb377cf4805629d0fb2acf while reworking to a cmake interface target ot-config. The author's suggestion is in  https://github.com/zephyrproject-rtos/openthread/commit/cb55c530bd54a795ef6989eee78cbe457c03d183#r39917273.

marvinroger, 2020-08-25:
The issue is not fixed, please remove the stale label.

markus-becker-tridonic-com, 2020-09-15:
It seems that upstream is deprecating NCP in favor of RCP:
https://groups.google.com/g/openthread-users/c/NsVMxljdUEo
https://github.com/openthread/ot-br-posix/pull/403

I have been testing otbr-agent from https://github.com/openthread/ot-br-posix and an RCP firmware from https://github.com/openthread/openthread on a NRF52840-DK without problems.

@rlubos, do you plan to change to RCP for NordicConnectSDK? Should we align in Zephyr?

markus-becker-tridonic-com, 2020-09-16:
I do have a working (but ugly) RCP configuration that talks to otbr-agent. Did not push yet, since it comments a lot of NCP code and replaces IS_ENABLED(CONFIG_OPENTHREAD_NCP) with IS_ENABLED(CONFIG_OPENTHREAD_RCP.

@rlubos would you want to keep the NCP alive or can it be removed completely and replaced by RCP since it is now the new default upstream?

rlubos, 2020-09-16:
@markus-becker-tridonic-com Thanks for taking the effort. I don't think we should remove the NCP support entirely (at least not at this point). The OpenThread doesn't seem to fully depreciate it yet (they're still shipping NCP sample in their repository) and there are users of this component (at least we use it for now in nRF Connect SDK). As for future plans for deprecation, I'd need to talk with my colleagues first, as they're more involved in the upstream OT development. 

Would it be possible to make RCP an option, so we still leave the users a choice? Would be good to see the actual changes though, to see how complex would it be.

markus-becker-tridonic-com, 2020-09-17:
@rlubos I have pushed the relevant changes. The pushes went to the Nordic fork of Zephyr. Since I don't know how the PRs would look like, if they would go to https://github.com/zephyrproject-rtos/zephyr. If the changes are acceptable, we can figure out how to upstream.

markus-becker-tridonic-com, 2020-09-17:
Would you have recommendations for the Kconfig to fix 

```
##[error]See https://docs.zephyrproject.org/latest/guides/kconfig/index.html for more details.

Found pointless 'menuconfig' symbols without children. Use regular 'config'
symbols instead. See
https://docs.zephyrproject.org/latest/guides/kconfig/tips.html#menuconfig-symbols.

OPENTHREAD_NCP                      subsys/net/l2/openthread/Kconfig:182
OPENTHREAD_RCP                      subsys/net/l2/openthread/Kconfig:190
```

rlubos, 2020-09-17:
It's happening, because `OPENTHREAD_NCP` was a `menuconfig` entry with some child entries, now it's no longer the case so now you should use the regular `config` for both `OPENTHREAD_NCP` and `OPENTHREAD_RCP`. 

Given that, probably we should rework the Kconfig structure for NCP/RCP, as I've now realilzed, we'll have ` OPENTHREAD_NCP` prefixed configs for RCP variant as well. An initial idea from my side would be to create a generic `menuconfig OPENTHREAD_COPROCESSOR`, and the put a `choice` entry underneath that would allow to select between NCP and RCP. Other related configs would go underneath `OPENTHREAD_COPROCESSOR` as well then. But I'm open for discussion here.

@edmont You've been dealing with OT Kconfigs recently, what do you think?

markus-becker-tridonic-com, 2020-09-17:
How about putting NCP as well as RCP in parallel to FTD/MTD?

rlubos, 2020-09-17:
This would work for RCP, but with NCP it's a bit more complex, as we still need to link either FTD or MTD libraries. So we would need to either fix to FTD for NCP (which impacts flexibility), or provide two different flavours of NCP config, NCP_FTD and NCP_MTD (which will make us put yet another conditional in the code).

markus-becker-tridonic-com, 2020-11-17:
Work on the PR that is fixing this issue continues.

closed 2020-11-19 by carlescufi

marvinroger, 2020-11-19:
Thanks to everyone involved!

PR #28458 (2020-11-19): net: openthread: RCP mode
OpenThread RCP mode integration into Zephyr

* Change RCP Kconfig option
* Limit OpenThread initialisation for RCP

Fixes #26383.