zephyrproject-rtos/zephyr #26391: Sample ubsys/usb/hid-cdc do not works

wysman, 2020-06-23:
The sample code can be compiled and flash into the target, but the linux host is unable to enumerate USB device.

Steps to reproduce the behavior:
1. west build -p auto -b stm32f746g_disco samples/subsys/usb/hid-cdc
2. west flash
3. journalctl -fk

**Expected behavior**
I expects to see some HID & UART popup.

**Impact**
The sample code can not be tested.

**Screenshots or console output**
```
usb 1-4: new full-speed USB device number 11 using xhci_hcd
usb 1-4: device descriptor read/64, error -32
usb 1-4: device descriptor read/64, error -32
usb 1-4: new full-speed USB device number 12 using xhci_hcd
usb 1-4: device descriptor read/64, error -32
usb 1-4: device descriptor read/64, error -32
usb usb1-port4: attempt power cycle
usb 1-4: new full-speed USB device number 13 using xhci_hcd
usb 1-4: Device not responding to setup address.
usb 1-4: Device not responding to setup address.
usb 1-4: device not accepting address 13, error -71
usb 1-4: new full-speed USB device number 14 using xhci_hcd
usb 1-4: Device not responding to setup address.
usb 1-4: Device not responding to setup address.
usb 1-4: device not accepting address 14, error -71
usb usb1-port4: unable to enumerate USB device
```

**Environment:**

```
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

```
Zephyr version: 2.3.99
Zephyr SDK version: 0.11.3
Board: stm32f746g_disco
```

**Working USB samples on the same dev kit**

- samples/subsys/usb/hid
- samples/subsys/usb/cdc_acm_composite
- samples/subsys/usb/mass

**Additional context**

```
r@cf3dd78b1cca:~/firmware/zephyr$ west build -p auto -b stm32f746g_disco samples/subsys/usb/hid-cdc
-- west build: making build dir /home/r/firmware/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/r/firmware/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/r/firmware/zephyr/samples/subsys/usb/hid-cdc
-- Zephyr version: 2.3.99 (/home/r/firmware/zephyr)
-- Found Python3: /usr/bin/python3.7 (found suitable exact version "3.7.3") found components:  Interpreter 
-- Board: stm32f746g_disco
ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found toolchain: zephyr (/opt/zephyr-sdk-0.11.3)
-- Found west: /usr/local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /opt/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/r/firmware/zephyr/boards/arm/stm32f746g_disco/stm32f746g_disco.dts
-- Generated zephyr.dts: /home/r/firmware/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/r/firmware/zephyr/build/zephyr/include/generated/devicetree_unfixed.h

warning: The choice symbol ENTROPY_DEVICE_RANDOM_GENERATOR (defined at subsys/random/Kconfig:37) was
selected (set =y), but TIMER_RANDOM_GENERATOR (defined at subsys/random/Kconfig:29) ended up as the
choice selection. See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_ENTROPY_DEVICE_RANDOM_GENERATOR.html
and/or look up ENTROPY_DEVICE_RANDOM_GENERATOR in the menuconfig/guiconfig interface. The
Application Development Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices
sections of the manual might be helpful too.

Parsing /home/r/firmware/zephyr/samples/subsys/usb/hid-cdc/Kconfig
Loaded configuration '/home/r/firmware/zephyr/boards/arm/stm32f746g_disco/stm32f746g_disco_defconfig'
Merged configuration '/home/r/firmware/zephyr/samples/subsys/usb/hid-cdc/prj.conf'
Configuration saved to '/home/r/firmware/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/r/firmware/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/r/.cache/zephyr
CMake Warning at ../../../../subsys/usb/CMakeLists.txt:28 (message):
  CONFIG_USB_DEVICE_VID has default value 0x2FE3.

  This value is only for testing and MUST be configured for USB products.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/r/firmware/zephyr/build
-- west build: building application
[1/145] Preparing syscall dependency handling

[140/145] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       45400 B         1 MB      4.33%
            DTCM:          0 GB        64 KB      0.00%
            SRAM:       21792 B       256 KB      8.31%
        IDT_LIST:         248 B         2 KB     12.11%
[145/145] Linking C executable zephyr/zephyr.elf
r@cf3dd78b1cca:~/firmware/zephyr$ west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/r/firmware/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-01341-g580d06d9d-dirty (2020-05-16-15:41)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J25M14 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.227858
Info : stm32f7x.cpu: hardware has 0 breakpoints, 10 watchpoints
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f7x.cpu       hla_target little stm32f7x.cpu       halted

Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080020d4 msp: 0x200114c0
Info : device id = 0x10016449
Info : flash size = 1024 kbytes
auto erase enabled
wrote 65536 bytes from file /home/r/firmware/zephyr/build/zephyr/zephyr.hex in 1.653791s (38.699 KiB/s)

Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080040bc msp: 0x20014340
verified 45400 bytes in 0.201106s (220.461 KiB/s)

Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
shutdown command invoked

```

jfischer-phytec-iot, 2020-06-24:
I guess it is the same as https://github.com/zephyrproject-rtos/zephyr/issues/25452#issuecomment-630903498, probably not enough endpoint to run the sample. 5[FS] IN / 5[FS] OUT

wysman, 2020-06-30:
Yes, it's the same issue.

emob-nordic, 2020-07-13:
Can we clsoe @wysman ?

wysman, 2020-07-16:
@emob-nordic Yes, we can close it. It's will be nice if "west/cmake" can detect and alert during build.

closed 2020-07-16 by wysman