zephyrproject-rtos/zephyr #26393: intc_loapic driver needs to be converted to DTS

andrewboie, 2020-06-23:
**Describe the bug**
The MMIO address for the LOAPIC is still coming from CONFIG_LOAPIC_BASE_ADDRESS. We're not supposed to be obtaining HW addresses from Kconfig any more.

Even though this is a per-cpu peripheral that is part of the processor core itself, the base address of the LOAPIC in the memory map should be conveyed via dts like other drivers. It seems IOAPIC has DTS bindings but not LOAPIC, probably an omission.

I have no strong opinions on how the ontology for this should be set up, whether LOAPIC/IOAPIC merged into one "APIC" peripheral or maintained separately, etc.

nisarumar, 2023-07-06:
Hi @nashif and @andrewboie
Looking at the current code base, seems like the issue is still not resolved. 
Is it okay if I work on this issue?
Thanks!

jhedberg, 2024-02-09:
This was already implemented through #60184

closed 2024-02-09 by jhedberg