zephyrproject-rtos/zephyr #26403: Compile Error when trying to build samples/synchronization

aaronhshi, 2020-06-24:
**Describe the bug**
Trying to build samples/synchronization but run into the following compilation errors.

**To Reproduce**
Steps to reproduce the behavior:
1. west build --pristine -b qemu_x86 
2. See error

**Expected behavior**
Clean build of sample project

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
\`\`\`aaronhshi@Aarons-MacBook-Pro synchronization % west build --pristine -b qemu_x86 
-- west build: making build dir /Users/aaronhshi/repo/testzephyrhello/zephyr/samples/synchronization/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /Users/aaronhshi/repo/testzephyrhello/zephyr/cmake/app/boilerplate.cmake
-- Application: /Users/aaronhshi/repo/testzephyrhello/zephyr/samples/synchronization
-- Zephyr version: 2.3.0 (/Users/aaronhshi/repo/testzephyrhello/zephyr)
-- Found Python3: /usr/local/bin/python3.7 (found suitable exact version "3.7.7") found components: Interpreter 
-- Board: qemu_x86
-- Found west: /usr/local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /usr/local/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /Users/aaronhshi/repo/testzephyrhello/zephyr/boards/x86/qemu_x86/qemu_x86.dts
-- Generated zephyr.dts: /Users/aaronhshi/repo/testzephyrhello/zephyr/samples/synchronization/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /Users/aaronhshi/repo/testzephyrhello/zephyr/samples/synchronization/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /Users/aaronhshi/repo/testzephyrhello/zephyr/Kconfig
Loaded configuration '/Users/aaronhshi/repo/testzephyrhello/zephyr/boards/x86/qemu_x86/qemu_x86_defconfig'
Merged configuration '/Users/aaronhshi/repo/testzephyrhello/zephyr/samples/synchronization/prj.conf'
Configuration saved to '/Users/aaronhshi/repo/testzephyrhello/zephyr/samples/synchronization/build/zephyr/.config'
Kconfig header saved to '/Users/aaronhshi/repo/testzephyrhello/zephyr/samples/synchronization/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- The ASM compiler identification is Clang
-- Found assembler: /Applications/Xcode.app/Contents/Developer/usr/bin/gcc
-- Cache files will be written to: /Users/aaronhshi/Library/Caches/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/aaronhshi/repo/testzephyrhello/zephyr/samples/synchronization/build
-- west build: building application
[1/113] Preparing syscall dependency handling

[7/113] Building C object zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj 
ccache /usr/bin/gcc -DBUILD_VERSION=zephyr-v2.3.0 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../../../kernel/include -I../../../arch/x86/include -I../../../include -Izephyr/include/generated -I../../../soc/x86/ia32 -isystem ../../../lib/libc/minimal/include -isystem /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/11.0.3/include -Os -imacros/Users/aaronhshi/repo/testzephyrhello/zephyr/samples/synchronization/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -imacros/Users/aaronhshi/repo/testzephyrhello/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -ffunction-sections -fdata-sections -mno-sse -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj   -c /Users/aaronhshi/repo/testzephyrhello/zephyr/arch/x86/core/offsets/offsets.c
In file included from /Users/aaronhshi/repo/testzephyrhello/zephyr/arch/x86/core/offsets/offsets.c:9:
../../../kernel/include/kernel_offsets.h:26:1: error: invalid operand in inline asm: '.globl	___cpu_t_current_OFFSET.equ	___cpu_t_current_OFFSET,${0:p}	.type	___cpu_t_current_OFFSET,@object'
GEN_OFFSET_SYM(_cpu_t, current);
^
../../../kernel/include/gen_offset.h:80:2: note: expanded from macro 'GEN_OFFSET_SYM'
        GEN_ABSOLUTE_SYM(__##S##_##M##_##OFFSET, offsetof(S, M))
        ^
../../../include/toolchain/gcc.h:405:10: note: expanded from macro 'GEN_ABSOLUTE_SYM'
        __asm__(".globl\t" #name "\n\t.equ\t" #name \
                ^
In file included from /Users/aaronhshi/repo/testzephyrhello/zephyr/arch/x86/core/offsets/offsets.c:9:
../../../kernel/include/kernel_offsets.h:26:1: error: unknown token in expression in '.equ' directive
../../../kernel/include/gen_offset.h:80:2: note: expanded from macro 'GEN_OFFSET_SYM'
        GEN_ABSOLUTE_SYM(__##S##_##M##_##OFFSET, offsetof(S, M))
        ^
../../../include/toolchain/gcc.h:405:30: note: expanded from macro 'GEN_ABSOLUTE_SYM'
        __asm__(".globl\t" #name "\n\t.equ\t" #name \
                                    ^
<inline asm>:2:31: note: instantiated into assembly here
        .equ    ___cpu_t_current_OFFSET,
                                        ^
In file included from /Users/aaronhshi/repo/testzephyrhello/zephyr/arch/x86/core/offsets/offsets.c:9:
[... 187 lines cut ...]
In file included from /Users/aaronhshi/repo/testzephyrhello/zephyr/arch/x86/core/offsets/offsets.c:9:
../../../kernel/include/kernel_offsets.h:41:1: error: missing expression in '.equ' directive
GEN_OFFSET_SYM(_ready_q_t, cache);
^
../../../kernel/include/gen_offset.h:80:2: note: expanded from macro 'GEN_OFFSET_SYM'
        GEN_ABSOLUTE_SYM(__##S##_##M##_##OFFSET, offsetof(S, M))
        ^
../../../include/toolchain/gcc.h:405:30: note: expanded from macro 'GEN_ABSOLUTE_SYM'
        __asm__(".globl\t" #name "\n\t.equ\t" #name \
                                    ^
<inline asm>:2:33: note: instantiated into assembly here
        .equ    ___ready_q_t_cache_OFFSET,
                                          ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /Users/aaronhshi/repo/testzephyrhello/zephyr/samples/synchronization/build\`\`\`

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: Host
 - Commit SHA or Version used: zephyr-v2.3.0 

**Additional context**
Add any other context about the problem here.

carlescufi, 2020-06-24:
>-- Found toolchain: host (gcc/ld)

You are trying to build for qemu_x86 on macOS. I do not think that is supported, because I am not sure that the default toolchain in macOS (LLVM/Clang) is actually compatible with Zephyr right now.

carlescufi, 2020-06-24:
Closing this, the combination is not supported on macOS.

closed 2020-06-24 by carlescufi