zephyrproject-rtos/zephyr #26406: On x86, the main stack overflows when CONFIG_NET_IPV6 and CONFIG_DEBUG are enabled

wobak-j, 2020-06-24:
**Describe the bug**
With IP v6 enabled, its initialization overflows the main stack in debug mode on qemu_x86.

**To Reproduce**
1. in _samples/net/socket/tcp/prj.conf_, add `CONFIG_DEBUG=y`
2. build _samples/net/socket/tcp_ with `-DBOARD=qemu_x86 -DOVERLAY_CONFIG=overlay-e1000.conf`
4. execute to produce the stack overflow 

**Expected behavior**
When IP v6 is enabled, the default stack size should be adjusted so that no overflow happens in debug mode.

**Environment (please complete the following information):**
 - OS: Ubuntu Linux
 - Zephyr SDK 0.11.3
 - Commit: 855b6c1a5ded091003d96a7b8d2b81662edbef70

jukkar, 2020-06-24:
The `CONFIG_DEBUG` is a kernel debugging option and not really related to IPv6 and networking. Please adjust the stack size manually in this case.

carlescufi, 2020-06-24:
@wobak-j having to adjust stack sizes based on different Kconfig options is expected. Please increase the stack size as required.

closed 2020-06-24 by carlescufi