zephyrproject-rtos/zephyr #26410: RFC: soc: Initial Nuvoton NPCX port

MulinChao, 2020-06-24:
## Introduction

This proposal plans to add initial support for Nuvoton NPCX series which is a family of highly integrated embedded controllers (EC) with an embedded ARM® Cortex®-M4 core and integrated advanced functions. It is targeted for a wide range of portable applications and provides complete EC functionality. 

### Problem description

Submitting the comments to add initial support for Nuvoton NPCX series via pull requests. 

### Proposed change

Supports Nuvoton NPCX soc and the npcx7m6fb EVB. 

## Detailed RFC

This proposal is expected to add:
 1. hal: add hal layer drivers for nuvoton npcx  series
 2. soc: introduce nuvoton npcx soc 
 3. driver: driver porting for UART and pinmux controller.
 4. board: add skeleton board support for npcx7m6fb_evb board from Nuvoton Technology.

### Proposed change (Detailed)

N/A

### Dependencies

N/A

### Concerns and Unresolved Questions

N/A

## Alternatives

N/A

MulinChao, 2020-06-24:
@jettr

carlescufi, 2020-06-24:
@MulinChao your plan sounds right. There is already a PR for another Nuvoton SoC family in the hal_nuvoton repo:
https://github.com/zephyrproject-rtos/hal_nuvoton/pull/1

jettr, 2020-06-25:
I didn't see any commits in the repository that you linked. I might be missing something though.

Since we are basing this support on the chromium ec counter part, I do think is still makes sense to add the npcx7/9 support to the main zephyr tree.

closed 2020-08-13 by ioannisg