zephyrproject-rtos/zephyr #26413: disco_l475_iot1: flash storage corruption caused by partition overlap

hwilmers, 2020-06-24:
**Describe the bug**

On disco_l475_iot1, the flash partition "storage" gets corrupted by mcuboot when an image swap is performed.

The cause of the problem is that in the devicetree for disco_l475_iot1, the flash partitions "storage" and "image-scratch"
have overlapping address spaces:
https://github.com/zephyrproject-rtos/zephyr/blob/979e124e21fb05cfedf8ed12078983d40e2f0b6c/boards/arm/disco_l475_iot1/disco_l475_iot1.dts#L173-L181

**To Reproduce**
Steps to reproduce the behavior:

1. build following sample: net/updatehub
2. perform an OTA firmware update
3. wait for system restart

**Expected behavior**

It is expected that after the firmware update, the original contents of partition "storage" are still intact.

**Impact**

This is preventing a usage of storage partition accross system updates.

**Environment (please complete the following information):**
 - Zephyr Version 2.3.0

hwilmers, 2020-06-24:
See #26417 for a proposed bugfix.

nandojve, 2020-06-24:
CC @mtahirbutt @otavio

PR #26417 (2020-07-01): boards: disco_l475_iot1: fix corruption of storage partition
Fixes #26413.

The devicetree for disco_l475_iot1 declares the flash partitions
"storage" and "image-scratch" with overlapping address spaces.

As a result, when one of these partitions is used it can corrupt the
contents of the other one. This is the case when mcuboot performs an
image swap using the scratch partition.

To fix the bug, the size of the scratch partition is reduced.
commit d09a2df5 (2020-07-01): boards: disco_l475_iot1: fix corruption of storage partition

Fixes #26413.

The devicetree for disco_l475_iot1 declares the flash partitions
"storage" and "image-scratch" with overlapping address spaces.

As a result, when one of these partitions is used it can corrupt the
contents of the other one. This is the case when mcuboot performs an
image swap using the scratch partition.

To fix the bug, the size of the scratch partition is reduced.
closed 2020-07-01 by nashif