zephyrproject-rtos/zephyr #26419: Cannot request update when writing to external flash

pdunaj, 2020-06-24:
**Describe the bug**
Nordic's qspi does not support writing if data comes from internal flash. The buffer must be in RAM.
https://github.com/zephyrproject-rtos/hal_nordic/blob/master/nrfx/drivers/src/nrfx_qspi.c#L355

When the image magic is written the buffer resides in flash.
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/dfu/boot/mcuboot.c#L100
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/dfu/boot/mcuboot.c#L386

**To Reproduce**
Steps to reproduce the behavior:
Create any setup where dfu is done through the external flash.

**Expected behavior**
Image magic should be writable.

**Impact**
Showstopper

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
master

**Additional context**
N/A

pdunaj, 2020-06-24:
@nvlsianpu

pdunaj, 2020-06-24:
It is enough to write to flash through the buffer in RAM. I copier the data to stack allocated array and performed write. Everything works. I think this is the best change to do so there will be no additional complexity done on the qspi driver side.

The other way to do it would be to handle it in the driver (detect that the buffer is not compatible and make a local copy in RAM). This may be more generic.

nvlsianpu, 2020-06-24:
I thought that https://github.com/zephyrproject-rtos/zephyr/pull/25971/ introduced support for from flash writes.
@pabigot

pabigot, 2020-06-24:
AFAIK #25971 works, but if the amount of data being written is 4 or more bytes you need to set CONFIG_NORDIC_QSPI_NOR_STACK_WRITE_BUFFER_SIZE=16` or whatever size buffer you want for the stack-based RAM region.  Only smaller values (which I know are used in MCUBoot) are handled without that option.

pdunaj, 2020-06-24:
I see. `CONFIG_NORDIC_QSPI_NOR_STACK_WRITE_BUFFER_SIZE` is not yet present downstream so I have not seen it.

pabigot, 2020-06-24:
It's a forward-port of an equivalent functionality that I thought made it into 1.3, but maybe it didn't.  @hakonfam

pdunaj, 2020-06-25:
There is `NORDIC_QSPI_NOR_FLASH_ALLOW_STACK_USAGE_FOR_DATA_IN_FLASH` in 1.3 indeed. I overlooked it.
When enabled magic is written correctly. Thanks @pabigot

closed 2020-06-25 by pdunaj