zephyrproject-rtos/zephyr #26427: Linker problems with zephyr-sdk-0.11.2

jaslob, 2020-06-24:
**Describe the bug**
I did a small application using time.h and receive an error when building the app.
main.c
#include <zephyr.h>
#include <stdio.h> 
#include <time.h> 

int main() 
{ 
	time_t now;
	now = time(0);
	printf("%s\n", ctime(&now));
	return 0; 
} 

prj.conf
CONFIG_NEWLIB_LIBC=y
CONFIG_MAIN_STACK_SIZE=2048

**To Reproduce**
 west build -p auto -b qemu_x86

**Screenshots or console output**
[3/10] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf 
: && /home/user/Zephyr/zephyr-sdk-0.11.2/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc -m32   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -Wl,-T  zephyr/linker.cmd  -Wl,-Map=/home/user/project/time/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/arch/x86/core/libarch__x86__core.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/x86/ia32/libsoc__x86__ia32.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/serial/libdrivers__serial.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/x86/core/offsets/offsets.c.obj  -L"/home/user/Zephyr/zephyr-sdk-0.11.2/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/32"  -L/home/user/project/time/build/zephyr  -lgcc  -Wl,--print-memory-usage  -m32  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lm  -lc  -lgcc  -L"/home/user/Zephyr/zephyr-sdk-0.11.2/x86_64-zephyr-elf/x86_64-zephyr-elf"/lib/32 && :
Memory region         Used Size  Region Size  %age Used
             RAM:      137528 B         4 MB      3.28%
        IDT_LIST:          80 B         2 KB      3.91/home/user/Zephyr/zephyr-sdk-0.11.2/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/../../../../x86_64-zephyr-elf/bin/ld: /home/user/Zephyr/zephyr-sdk-0.11.2/x86_64-zephyr-elf/x86_64-zephyr-elf/lib/32/libc.a(lib_a-gettimeofdayr.o): in function `_gettimeofday_r':
gettimeofdayr.c:(.text._gettimeofday_r+0x1a): undefined reference to `gettimeofday'
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/user/project/time/build

**Environment (please complete the following information):**
 - OS: Clear Linux
 - Toolchain zephyr-sdk-0.11.2

dcpleung, 2020-06-24:
This is fixed by commit cd1657df97b21b34c7051211dac3132b9ef3e67d which is after 0.11.3. Once we have a new SDK release, this should not be an issue anymore.

galak, 2020-06-24:
Cool, looking to make a 0.11.4 soon w/that fix, the march=pentium and a packaging fix.

jaslob, 2020-06-25:
Ok thanks. I will get that update when it's released.

pabigot, 2020-06-29:
@jaslob https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.11.4 has the updated SDK.  Can you confirm the problem is resolved with it?  Thanks.

galak, 2020-07-16:
Closing this as v0.11.4 should fix the issue.  Feel free to re-open if it doesn't.

closed 2020-07-16 by galak