zephyrproject-rtos/zephyr #26434: nrf9160 uart_tx can return -ENOTSUP, which is not documented behavior

jwise, 2020-06-25:
**Describe the bug**
On nRF9160, the UART driver when running in async mode can return `-ENOTSUP` when trying to send something that is not in RAM.  However, this is not documented in https://elixir.bootlin.com/zephyr/latest/source/include/drivers/uart.h#L449 . 

**To Reproduce**
Steps to reproduce the behavior:
1. Call `uart_tx(uart, "string in flash", 15, ...)` from user program.

**Expected behavior**
The driver should copy data into RAM (seems unlikely), the hardware should be ECOed to support DMA from flash (extra unlikely :) ), or the documentation should be updated to reflect that `-ENOTSUP` is a possible outcome.

**Impact**
This is mostly an annoyance (it cost me about a half hour of debugging).

jfischer-phytec-iot, 2020-06-25:
>However, this is not documented

There is unambiguous description:
`retval 0	  If successful, negative errno code otherwise.`

jwise, 2020-06-25:
Hmm, I disagree.  It gives the possibilities as `EBUSY` and `0`:

```
 *
 * @retval -EBUSY There is already an ongoing transfer.
 * @retval 0	  If successful, negative errno code otherwise.
 */
```

`ENOTSUP` for sending a const buffer really seems like the sort of thing that ought be documented.

pabigot, 2020-06-29:
Agreed; that one negative value is documented, but others are obscured by being identified in [the documentation](https://docs.zephyrproject.org/apidoc/latest/group__uart__interface_ga281af2ea0253d2b94b01fca4f0932cc1.html#ga281af2ea0253d2b94b01fca4f0932cc1) for the case that returns `0` is not ideal.

I like:
```
@retval 0 on success
@retval -EBUSY if a transfer is already in progress
@retval negative for other error cases
```
to describe the circumstances under which a particular value might occur while not excluding the possibility of other errors values.

Note that multiple API functions in this module use the obscure idiom.

closed 2020-09-12 by github-actions[bot]