zephyrproject-rtos/zephyr #26435: Suspicious source code with subsys/random/random32_entropy_device: seg fault risk

elsinkior, 2020-06-25:
**Describe the bug**
In the source code subsys/random/random32_entropy_device, static function rand_get: risk of seg fault if output requested length is not modulo 4.

**To Reproduce**
NA
**Expected behavior**
Shall be fixed to avoid seg fault with length % 4 != 0 as input

**Impact**
NA

**Screenshots or console output**
NA

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain : zephyr-sdk-0.11.3
 - Commit SHA 979e124e21fb05cfedf8ed12078983d40e2f0b6c

**Additional context**
NA

pabigot, 2020-06-29:
I don't think there's a problem if `outlen` isn't a multiple of 4: the branch to use `memcpy` to cover the final bytes should handle it.

But the assignment through `(uint32_t*)&dst[len]` is incorrect and likely to cause a fault if `dst` is not a pointer to a 4-byte aligned address.

The body of the loop should just use `memcpy` for however many bytes it stores; trust the compiler to optimize the copy.

ceolin, 2020-06-29:
actually the `memcpy` is wrong too:

```
				(void)memcpy(&(dst[random_num]),
						&random_num, blocksize);
```

where `random_num` is `random_num = k_cycle_get_32();`

PR #26550 (2020-07-01): random: entropy: Fix invalid memory access
In sys_rand_get() if the entropy hardware unlikely return error, the
fallgback is using the system timer to and fill the given buffer with
that data.

The problem what that k_cycle_get_32() returns a four bytes integer and
depending the sizeof of the buffer we need copy the right amount of
data. That was not being done properly leading to invalid read/write
memory access.

Fixes #26435
commit 12508024 (2020-07-01): random: entropy: Fix invalid memory access

In sys_rand_get() if the entropy hardware unlikely return error, the
fallgback is using the system timer to and fill the given buffer with
that data.

The problem what that k_cycle_get_32() returns a four bytes integer and
depending the sizeof of the buffer we need copy the right amount of
data. That was not being done properly leading to invalid read/write
memory access.

Fixes #26435
closed 2020-07-01 by nashif