zephyrproject-rtos/zephyr #26438: Bluetooth: Reconnection to paired/bonded peripheral fails

jsiverskog, 2020-06-25:
**Describe the bug**
I'm having an nrf52 Zephyr v2.3.0 system acting as central, connecting/pairing/bonding with an nrf51 peripheral running SoftDevice 7.0.0. Since the peripheral has already been on the market for years, there's not much I can do on that side. It's working well with other stacks (BlueZ, iOS/macOS/Android).

In some cases the connection/pairing procedure fails.

**To Reproduce**
Central: nrf52832 running zephyr v2.3.0, mac da:04:23:a7:cc:a8
Peripheral: nrf51 running softdevice 7.0.0, macs d9:25:8b:f6:89:de and dd:47:38:ae:75:57

Steps to reproduce the behavior:
1. peripheral: activate advertising (mac d9:25:8b:f6:89:de)
2. central: activate scanning
3. central: connect and pair with peripheral (when found) [debug_log.txt:17] [device_b.pcapng:8962]
4. central: subscribe to services
5. central: receive notifications [debug_log.txt:353] [device_b.pcapng:9225]

6. peripheral: remove batteries (to emulate disconnect) [debug_log.txt:961]
7. peripheral: insert batteries (will advertise with same mac address as before, pairing/bonding information should be restored)
8. central: activate scanning
9. central: reconnect to peripheral [debug_log.txt:984] [device_b.pcapng:10662]
10. central: receive notifications [debug_log.txt:1206] [device_b.pcapng:10798]

[The above sequence can be followed in device_a.pcapng]

11. peripheral: reset device (clear pairing/bonding info, start up with new mac address dd:47:38:ae:75:57) [debug_log.txt:1574]
12. peripheral: activate advertising with new mac
13. central: activate scanning
14. central: connect to peripheral [debug_log.txt:1597] [device_b.pcapng:6397]
15. central: receive notifications [debug_log.txt:1954] [device_b.pcapng:6540]

16. peripheral: remove batteries (to emulate disconnect) [debug_log.txt:2354]
17. peripheral: insert batteries (will advertise with same mac address as before, pairing/bonding information should be restored)
18. central: activate scanning
19. central: reconnect to peripheral [debug_log.txt:2376] [device_b.pcapng:8079]

At this point the two peers will end up in a connect/disconnect loop which they
don't seem to be able to recover from. Since it works the first time, the nrf51
device resets when changing mac address, and it works if I reflash the nrf52
device, I get the feeling that it's something on the nrf52 side.

[The above sequence can be followed in device_b.pcapng]

**Expected behavior**
Connection successful and data transferred.

Attached: debug log from the central as well as Wireshark sniffs.

[debug_log.txt](https://github.com/zephyrproject-rtos/zephyr/files/4830234/debug_log.txt)
[sniffs.zip](https://github.com/zephyrproject-rtos/zephyr/files/4830235/sniffs.zip)

jsiverskog, 2020-06-25:
note that this is 100% reproducible - it always works for the first connection (and subsequent reconnects), and _always_ fails on reconnects to the second connection.

jsiverskog, 2020-06-29:
i *might* have a workaround - if i before doing the subsequent connect call `bt_unpair`, it seems to work. does that retrigger key negotiation?

carlescufi, 2020-07-08:
@jsiverskog could it be that you have unauth repairing disabled?
There's a KConfig option for this: `CONFIG_BT_SMP_ALLOW_UNAUTH_OVERWRITE`. Try enabling that and see if it fixes your issue.

joerchan, 2020-07-13:
> i have found a workaround - if i before doing the subsequent connect call `bt_unpair`, it seems to work. does that retrigger key negotiation?

It would do a new pairing+bonding procedure, and new keys would be distributed.

I cannot tell from the sniffer logs and debug logs what is the issue here.
The slave dissappears from the connection (in device_b wireshark log), and I cannot see why that happened.
Is it possible to get the disconnect reason from the peripheral side? 

Does it always happen to the same device? Or is it the device that connects last?

jsiverskog, 2020-07-14:
> It would do a new pairing+bonding procedure, and new keys would be distributed.
>
> I cannot tell from the sniffer logs and debug logs what is the issue here.
> The slave dissappears from the connection [...]

unfortunately it's hard to get any logs out of it.

> 
> Does it always happen to the same device? Or is it the device that connects last?

i did try another case now, which indicated that it's *not* on the zephyr side. instead of resetting the peripheral, i tried with a physically different peripheral, and it seems to work. i can't believe why i didn't try this earlier, but that is a sign something is not reset / handled correctly on the peripheral side.

> 
> > There's a Kconfig option for this: CONFIG_BT_SMP_ALLOW_UNAUTH_OVERWRITE. Try enabling that and see if it fixes your issue.
> 
> From the sniffer log this does not appear to be the problem, and enabling the option would not have an impact on the behavior

i tried enabling it but it doesn't seem to make a difference.

joerchan, 2020-07-14:
@jsiverskog I would agree with you that this appears to be an issue on the peripheral. I cannot see that this has anything to do with the central/zephyr. Also no clue on what happened on the peripheral side.

I will close this issue since it is not a zephyr problem.

closed 2020-07-14 by joerchan