zephyrproject-rtos/zephyr #26450: Bad disconnect reason when client connects with wrong address type

Emil-Gydesen-Bose, 2020-06-25:
**Describe the bug**
If a client connects with a wrong address type (i.e. public instead of random), then the disconnect reason from the server is `BT_HCI_ERR_MEM_CAPACITY_EXCEEDED`.

**To Reproduce**
Using the BT Shell application:

Advertiser:
```
bt init
bt advertise on
```
Which will start advertising with a random (resolvable address)

Initiator:
```
bt init
bt connect <address> public
```
where `address` is the random resolvable address from the advertiser. 

**Expected behavior**
Expect to see some other error code (e.g. `BT_HCI_ERR_BD_ADDR_UNACCEPTABLE` or `BT_HCI_ERR_UNACCEPT_CONN_PARAM`), or perhaps something entirely different should happen. 

**Impact**
Low impact as this is a (human) client error. 

**Logs and console output**
BTMON log from the server
```
> HCI Event: LE Meta Event (0x3e) plen 31                                                     #210 [hci1] 490.763947
      LE Enhanced Connection Complete (0x0a)
        Status: Success (0x00)
        Handle: 3585
        Role: Master (0x00)
        Peer address type: Random (0x01)
        Peer address: 47:BE:39:10:67:B7 (Resolvable)
        Local resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Peer resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Connection interval: 45.00 msec (0x0024)
        Connection latency: 0 (0x0000)
        Supervision timeout: 4000 msec (0x0190)
        Master clock accuracy: 0x00
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                #211 [hci1] 490.802547
        Handle: 3585
        Reason: Memory Capacity Exceeded (0x07)
```

**Environment (please complete the following information):**
Using Linux with the `native_posix` board and commit `af67b297a38a31815f4adeea0fab215858203c1c` with the `/tests/bluetooth/shell` project. 

**Additional context**
Comment from @joerchan :
> The reason for the disconnect is that the Host does not have enough context's for the new connection. Which is a situation we should never find ourselves in.

joerchan, 2020-06-26:
From the BTMON log this is clearly a bug in the controller, the address type was not used for address match by the initiator.
```
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 26                                     #47 38.778296
        Peer address type: Public (0x00)
        Peer address: 50:D5:B6:53:75:BD (OUI 50-D5-B6)
> HCI Event: LE Meta Event (0x3e) plen 31                                                              #51 38.810379
      LE Enhanced Connection Complete (0x0a)
        Peer address type: Random (0x01)
        Peer address: 50:D5:B6:53:75:BD (Resolvable)
```

closed 2020-07-06 by carlescufi

Emil-Gydesen-Bose, 2020-07-06:
> From the BTMON log this is clearly a bug in the controller, the address type was not used for address match by the initiator.
>
> ```
> < HCI Command: LE Extended Create Connection (0x08|0x0043) plen [...]

Just a FYI: The implementors of the controller have been notified about this issue.

PR #26465 (2020-07-06): Bluetooth: host: Unspecified disconnect reason for no pending conn
Use an unspecified reason when disconnecting a connection that did not
have a pending connection in the host.
We don't allocate connection objects in the connection complete anymore
so using memory-capacity-exceeded is misleading.

Fixes: #26450