zephyrproject-rtos/zephyr #26454: k_delayed_work queues and user mode

andrewboie, 2020-06-25:
**Is your enhancement proposal related to a problem? Please describe.**
At the moment, none of the k_delayed_work APIs work from user mode.

It has been quite some time since I converted the IPC objects to be usable from user mode, and I don't recall the exact reason why I did not convert k_delayed_work APIs, but I'm pretty sure it had something to do with the `struct k_delayed_work` object and security concerns.

**Describe the solution you'd like**
A mechanism that meets the current requirements of k_delayed_work, that user mode threads can use.

I would prefer a solution that minimizes adding new syscalls, and just uses existing kernel object syscalls as much as possible.

It would be really good if work could be submitted to these queues from an ISR. Otherwise this wouldn't need to be part of the core kernel and could just be library code/an exercise for the end user.

**Describe alternatives you've considered**
Some redesign of the APIs could be on the table. Depending on the problems we face, a parallel API set like we did with sys_sem/k_sem and sys_mem_pool/k_mem_pool might be acceptable (the whole thing could be futex-based in that case)

dcpleung, 2021-03-10:
@pabigot is this still relevant after your work queue rework?

pabigot, 2021-03-11:
We don't have a good solution or clear motivation for userspace work queues (with requirements and stakeholder expectations), but the API or at least implementation that's used in supervisor mode isn't one of them.  That's never going to work in usermode, and there's no clear reason why it should.

So the answer is either "yes" if people want userspace work queues, or "no" if people don't care about them.

dcpleung, 2024-02-08:
I guess no one cares about this for now... reopen if needed.

closed 2024-02-08 by dcpleung