zephyrproject-rtos/zephyr #26455: bme280 connect to rt1020_evk

stratonflorin, 2020-06-25:
device_get_binding(BME280_LABEL); return 0

MaureenHelm, 2020-06-25:
Hi @stratonflorin, please provide more details about your issue. Which Zephyr version (git tag or SHA) did you use? How did you attach a bme280 to this board? Is this an in-tree sample application or a custom application?

stratonflorin, 2020-06-25:
Hi!
I am using Zephyr SDK 0.11.3

This is the connection scheme i use (GPIO_SD_B1_03 for I2C_SDA and GPIO_SD_B1_02 for I2C_SCL, both from J19 area).
And the aplication is in-tree: samples/sensor/bme280.

I created an overlay for mimxrt1020_evk which looks like this

```
&lpi2c4 {
	bme280@76 {
		compatible = "bosch,bme280";
		reg = <0x76>;
		label = "BME280";
	};
};

```
And the problem i have is that the function `device_get_binding(BME280_LABEL);` returns a null structure.
BME280_LABEL has value "BME280".

[image]

MaureenHelm, 2020-06-25:
Thanks for the additional information. It looks like you've configured your dts and wired the boards correctly. I suspect the sensor initialization may be failing in `bme280_init()`. Can you increase the log level by setting `CONFIG_SENSOR_LOG_LEVEL_DBG=y` in the application `prj.conf`, then share the console output?

cc: @mbolivar-nordic in case 8aadf8957989ecbe10bcd632fe49ec035a8b536f might be to blame.

stratonflorin, 2020-06-25:
```
west build -p auto -b mimxrt1020_evk samples/sensor/bme280
[0/1] Re-running CMake...
Including boilerplate (Zephyr base (cached)): /home/sfc/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/sfc/zephyrproject/zephyr/samples/sensor/bme280
-- Zephyr version: 2.3.0-rc1 (/home/sfc/zephyrproject/zephyr)
-- Board: mimxrt1020_evk
-- Found toolchain: zephyr (/home/sfc/zephyr-sdk-0.11.3)
-- Found west: /home/sfc/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /home/sfc/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/sfc/zephyrproject/zephyr/boards/arm/mimxrt1020_evk/mimxrt1020_evk.dts
-- Found devicetree overlay: /home/sfc/zephyrproject/zephyr/samples/sensor/bme280/boards/mimxrt1020_evk.overlay
-- Generated zephyr.dts: /home/sfc/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/sfc/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/sfc/zephyrproject/zephyr/samples/sensor/bme280/Kconfig
Loaded configuration '/home/sfc/zephyrproject/zephyr/boards/arm/mimxrt1020_evk/mimxrt1020_evk_defconfig'
Merged configuration '/home/sfc/zephyrproject/zephyr/samples/sensor/bme280/prj.conf'
Configuration saved to '/home/sfc/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/sfc/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Cache files will be written to: /home/sfc/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sfc/zephyrproject/zephyr/build
[126/131] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           OCRAM:          0 GB       128 KB      0.00%
            DTCM:          0 GB        64 KB      0.00%
            ITCM:          0 GB        64 KB      0.00%
           FLASH:       34908 B         8 MB      0.42%
            SRAM:        6504 B        32 MB      0.02%
        IDT_LIST:         200 B         2 KB      9.77%
[131/131] Linking C executable zephyr/zephyr.elf
```

```
west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner jlink
-- runners.jlink: Flashing file: /home/sfc/zephyrproject/zephyr/build/zephyr/zephyr.bin
SEGGER J-Link Commander V6.80a (Compiled May 29 2020 16:30:49)
DLL version V6.80a, compiled May 29 2020 16:30:38

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link OpenSDA 2 compiled May 27 2019 11:01:03
Hardware version: V1.00
S/N: 621000000
VTref=3.300V
Target connection not established yet but required for command.
Device "MIMXRT1021XXX5A" selected.

Connecting to target via SWD
Found SW-DP with ID 0x0BD11477
Unknown DP version. Assuming DPv0
Scanning AP map to find all available APs
AP[1]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x04770041)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FD000
CPUID register: 0x411FC271. Implementer code: 0x41 (ARM)
Found Cortex-M7 r1p1, Little endian.
FPUnit: 8 code (BP) slots and 0 literal slots
CoreSight components:
ROMTbl[0] @ E00FD000
ROMTbl[0][0]: E00FE000, CID: B105100D, PID: 000BB4C8 ROM Table
ROMTbl[1] @ E00FE000
ROMTbl[1][0]: E00FF000, CID: B105100D, PID: 000BB4C7 ROM Table
ROMTbl[2] @ E00FF000
ROMTbl[2][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
ROMTbl[2][1]: E0001000, CID: B105E00D, PID: 000BB002 DWT
ROMTbl[2][2]: E0002000, CID: B105E00D, PID: 000BB00E FPB-M7
ROMTbl[2][3]: E0000000, CID: B105E00D, PID: 000BB001 ITM
ROMTbl[1][1]: E0041000, CID: B105900D, PID: 001BB975 ETM-M7
ROMTbl[1][2]: E0042000, CID: B105900D, PID: 004BB906 CTI
ROMTbl[0][1]: E0040000, CID: B105900D, PID: 000BB9A9 TPIU-M7
ROMTbl[0][2]: E0043000, CID: B105F00D, PID: 001BB101 TSG
Cache: Separate I- and D-cache.
I-Cache L1: 16 KB, 256 Sets, 32 Bytes/Line, 2-Way
D-Cache L1: 16 KB, 128 Sets, 32 Bytes/Line, 4-Way
Cortex-M7 identified.
Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.

Downloading file [/home/sfc/zephyrproject/zephyr/build/zephyr/zephyr.bin]...
Comparing flash   [100%] Done.
J-Link: Flash download: Bank 0 @ 0x60000000: Skipped. Contents already match
O.K.

Script processing completed.
```

```
[00:00:00.000,000] <dbg> BME280.bme280_init: initializing BME280
[00:00:00.000,000] <dbg> BME280.bme280_chip_init: ID read failed: -5
[00:00:00.000,000] <dbg> BME280.bme280_init: BME280 failed
*** Booting Zephyr OS build v2.3.0-rc1-384-g8b5b7fcf8d84  ***
No device "BME280" found; did initialization fail?

```

stratonflorin, 2020-06-26:
hi, @MaureenHelm!
I'm sorry I'm so insistent. Do you have any advice on what to do next?

MaureenHelm, 2020-06-26:
Try attaching a debugger and stepping through `bme280_chip_init()`. You may also want to attach a logic analyzer to the I2C pins to see what's happening on the bus.

stratonflorin, 2020-06-26:
i don'a have a logic analizer for debugging
can you mail me at straton.florin.c@gmail.com?

mbolivar-nordic, 2020-06-26:
> cc: @mbolivar-nordic in case [8aadf89](https://github.com/zephyrproject-rtos/zephyr/commit/8aadf8957989ecbe10bcd632fe49ec035a8b536f) might be to blame.

I retested the sample on nRF52840 on master (fdab8f189d) and v2.3.0, no issues connecting via I2C. Sample worked out of the box on `nrf52840dk_nrf52840` using the upstream overlay.

>  I am using Zephyr SDK 0.11.3

That's not a Zephyr version; the SDK is an independently versioned piece of software. Your zephyr version is what you get when you run `git describe` in the zephyr repository (make sure to run `west update` and do a pristine build).

mbolivar-nordic, 2020-06-26:
```
		reg = <0x76>;
```

@stratonflorin did you get this right? The sensor has a strap pin you can use to customize the address.

stratonflorin, 2020-06-26:
v2.3.0-rc1-384-g8b5b7fcf8d is my current version

yes, it has SDO pin for changing address. mine is connected to GND so this is the address
nothing changed after running west update and rebuilding application

still this output

```
[00:00:00.000,000] <dbg> BME280.bme280_init: initializing BME280
[00:00:00.000,000] <dbg> BME280.bme280_chip_init: ID read failed: -5
[00:00:00.000,000] <dbg> BME280.bme280_init: BME280 failed
*** Booting Zephyr OS build v2.3.0-rc1-384-g8b5b7fcf8d84  ***
No device "BME280" found; did initialization fail?
```

mbolivar-nordic, 2020-06-27:
> v2.3.0-rc1-384-g8b5b7fcf8d is my current version

That's kind of a "random" commit from the v2.3.0 RC series -- testing with the final v2.3.0 release or master is probably best if you can. But I know that can be hard to do depending on what you've got committed on top.

> yes, it has SDO pin for changing address. mine is connected to GND so this is the address [...]
> still this output

Since you don't have a logic analyzer, you'll probably have to enable debug logs in your I2C driver or add some if the right ones aren't present.

From the logs, it looks like i2c_burst_read() is failing when called by bme280_reg_read_i2c():

https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/sensor/bme280/bme280.c#L460

Good luck!

stratonflorin, 2020-06-27:
i don't think i understand what i have to do

mbolivar-nordic, 2020-06-27:
I changed the issue type from "bug" to "question" since that seems more appropriate at this time.

I'm sorry I won't be able to help further, but to me it looks like the BME sample application and driver are both working correctly, so I'm unsubscribing from this issue for now. Please feel free to ping me if there's a bug.

stratonflorin, 2020-06-27:
@mbolivar-nordic, i understand
can you mail me please at straton.florin.c@gmail.com?

MaureenHelm, 2020-07-01:
> > cc: @mbolivar-nordic in case [8aadf89](https://github.com/zephyrproject-rtos/zephyr/commit/8aadf8957989ecbe10bcd632fe49ec035a8b536f) might be to blame.
> 
> I retested the sample on nRF52840 on master ([fdab8f1](https://github.com/zephyrproject-rtos/zephyr/commit/fdab8f189dd70b979bd1622004f852854c235ce5)) and v2.3.0, no issues connecting via I2C. Sample worked out of the box on `nrf52840dk_nrf52840` using the upstream overlay.

Thanks for verifying.

I still recommend that you try using the on-board debug probe to step through `bme280_chip_init()`. You can find more information about how to use the probe [here](https://docs.zephyrproject.org/latest/boards/arm/mimxrt1020_evk/doc/index.html#programming-and-debugging).

> Since you don't have a logic analyzer, you'll probably have to enable debug logs in your I2C driver or add some if the right ones aren't present.

This means you should add LOG_DBG or LOG_ERR statements to `drivers/i2c/i2c_mcux_lpi2c.c`. See `drivers/spi/spi_mcux_lpspi.c` for example.

closed 2020-09-21 by github-actions[bot]