zephyrproject-rtos/zephyr #26466: Bluetooth: host: Do auto-postponement of advertising also when application requests advertising

wdhpawa, 2020-06-26:
**Is your enhancement proposal related to a problem? Please describe.**
If the application layer successfully starts advertising by calling bt_le_adv_start without the BT_LE_ADV_OPT_ONE_TIME option, the Zephyr host layer will automatically restart advertising when it stops due to a new connection. If there is no room for more connections at this point, the host will postpone the advertising until there is a disconnect, at which time the advertising will be started automatically. Thus, the application only needs to make the first start request, then all subsequent stopping/starting is handled automatically by the host.

However, if the application explicitly stops and starts advertising for other reasons (e.g. due to changing the advertising interval or whitelist, or performing some other activity that requires advertising to be stopped), and the application's start request comes at a time when there is no room for more connections, the Zephyr host will simply reject the request and do nothing further. This means that it is now up to the application to handle the same problem that is normally handled by the host, i.e. the application must postpone the advertising until there is a disconnect, and then explicitly start advertising at that point.

The fact that the application is forced to handle this advertising postponement in some cases, means that there is no real advantage to letting the Zephyr host handle it at all. We have therefore, for now, reimplemented the feature in the application and disabled it in the host by setting the BT_LE_ADV_OPT_ONE_TIME option.

**Describe the solution you'd like**
Our proposal is that the Zephyr host should do the same automatic postponement of advertising when the application explicitly calls bt_le_adv_start - if the BT_LE_ADV_OPT_ONE_TIME option is not set, and there is no room for more connections. This would make the postponement feature useful, and would completely eliminate the need for the application to deal with the limit on the number of connections. See the attached (simplified) sequence diagram.
[image]

**Describe alternatives you've considered**
The obvious alternative is to reimplement the feature in the application, but that renders the existing feature in the stack useless.

**Additional context**
None.

carlescufi, 2020-07-08:
@wdhpawa Thanks for the report, this makes quite a lot of sense. Would you mind proposing a patch yourself?

wdhpawa, 2020-07-09:
@carlescufi I'm afraid it doesn't look like we'll be able to prioritize this at our end due to an approaching deadline, so we're going to have to make do with the mentioned workaround for now. Our intention was just to create awareness of the problem, so maybe someone would have time to fix it eventually.

PR #26861 (2020-07-15): Bluetooth: host: Start persistent advertiser with max conn reached
Allow calling bt_le_adv_start with no connection object available
when not the option BT_LE_ADV_OPT_ONE_TIME has been set. In this
case the advertiser will be restarted once there is a connectable
advertiser object available.

Fixes: #26466
closed 2020-07-15 by carlescufi