zephyrproject-rtos/zephyr #26488: Bluetooth: Connection failure using frdm_kw41z shield

MaureenHelm, 2020-06-26:
**Describe the bug**
The `frdm_kw41z` when used Bluetooth controller shield attached to `frdm_k64f`, `mimxrt1050_evk`, and other boards fails to make a Bluetooth connection.

**To Reproduce**
1. Build the Bluetooth controller for `frdm_kw41z` as described [here](https://docs.zephyrproject.org/latest/boards/shields/frdm_kw41z/doc/index.html) and load it to the `frdm_kw41z` flash.
2. Build the peripheral_hr sample for `frdm_k64f` and load it to the `frdm_k64f` flash
   ```
   $ west build -b frdm_k64f -p auto samples/bluetooth/peripheral_hr/ -- -DSHIELD=frdm_kw41z
   $ west flash
   ```
3. Use NXP IoT Toolbox or nRF Connect app on an iPhone and attempt to connect. The connection fails and errors are logged to the console.

**Expected behavior**
The Bluetooth connection succeeds and heart rate data is transmitted to the phone app.

**Impact**
`frdm_kw41z` cannot be used as a Bluetooth controller shield

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.3.0-608-g22be0d8fd590  ***
Bluetooth initialized
Advertising successfully started
[00:00:01.039,000] <wrn> bt_hci_core: Controller to host flow control not supported
[00:00:01.055,000] <inf> bt_hci_core: Identity: 00:60:37:16:d9:d1 (public)
[00:00:01.055,000] <inf> bt_hci_core: HCI: version 4.2 (0x08) revision 0x000b, manufacturer 0x0025
[00:00:01.055,000] <inf> bt_hci_core: LMP: version 4.2 (0x08) subver 0x0123
[00:00:25.875,000] <err> bt_hci_core: Unable to lookup conn for handle 32
[00:00:26.084,000] <err> bt_hci_core: Unable to lookup conn for handle 32
[00:00:26.114,000] <err> bt_hci_core: Unable to find conn for handle 32
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.11.3
 - Commit 22be0d8fd5902da11c9bf55a44012abc7c2cf1c1

**Additional context**
This worked in the past, but I haven't yet been able to isolate when it broke.

dleach02, 2020-07-06:
There is a known bug in the KW41z code that has been fixed. The release is pending. When released, this issue report will be updated with the details and then closed.

dleach02, 2020-09-15:
I have confirmed that this problem is fixed with the pending KW41 SDK 2.2.3 release. The current schedule for website release of the SDK 2.2.3 is October 1, 2020.

dleach02, 2020-10-06:
Release of KW41Z SDK 2.2.3 fixes this. Will update documentation to ensure folks grab this release if trying to use the frdm_kw41z shield for BLE

closed 2020-10-07 by dleach02