zephyrproject-rtos/zephyr #26500: sanitycheck reports failing tests with test_slice_scheduling()

yingmingx, 2020-06-27:
**Describe the bug**
When I was optimize my PR #25813, there is always a failure on test_slice_scheduling.
Handler.log shows all other platforms are good except qemu_arc_hs and qemu_arc_em.
So I checkout git branch to master and change macro ITRERATION_COUNT in  test_priority_scheduling.c and
test_slice_scheduling.c from 5 to 10, this happens, but not every time.

**To Reproduce**
Steps to reproduce the behavior:
1. In master branch, commit ID ba0a791635c9fa70f0dc5ffb7a98a03f7e28f000
2. change macro ITRERATION_COUNT in  test_priority_scheduling.c and test_slice_scheduling.c from 5 to 10
3. run command: sanitycheck -T tests/kernel/sched/schedule_api/ 

**Logs and console output**
START - test_slice_reset

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:91: thread_time_slice: (t <= expected_slice_max is false)
timeslice too big, expected 110000 got 115322

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:88: thread_time_slice: (t >= expected_slice_min is false)
timeslice too small, expected 90000 got 15903

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:88: thread_time_slice: (t >= expected_slice_min is false)
timeslice too small, expected 90000 got 960
 FAIL - test_slice_reset
===================================================================
START - test_slice_scheduling
AB
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

C
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

D
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

E
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

F
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

G
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

H
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

I
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

A
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

 FAIL - test_slice_scheduling
===================================================================
START - test_priority_scheduling
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
 PASS - test_priority_scheduling

jenmwms, 2020-06-30:
@yingmingx  Thanks for the info. Can you please clarify - is this issue observed on master (referenced in description) or with your PR, or both? I will try locally too

yingmingx, 2020-07-01:
@jenmwms This issue observed not only on my PR, but also on the master.

abrodkin, 2020-08-18:
@galak pinging you here as well - what about new SDK with QEMU with `icount` for ARC :)

andrewboie, 2020-09-23:
This is a known issue with any QEMU where icount isn't enabled. This does not affect real hardware, and we have alternative emulators like nsim free which we can use for validation instead. this will sort itself out once the new SDK lands.

@MaureenHelm I recommend we downgrade this to low since it isn't a HW problem, just QEMU.

closed 2020-11-23 by yingmingx