zephyrproject-rtos/zephyr #26509: net_l2_ppp.ppp_link_terminated: SARA U201 modem

mtahirbutt, 2020-06-28:
**Describe the bug**
A clear and concise description of what the bug is.
I am trying to connect SARA U201 modem via PPP link. but sometimes get net_l2_ppp.ppp_link_terminated issue so not able to send data via zephyr TCP/IP stack. I am trying to connect via UDP updatehub app.  Sometimes, I am successful in making the connection but mostly I get this issue PPP link termination issue. Therefore, right now not able to get sustained performance.
Sometimes I see no reply for Sending Configure-Req messages, but the most problematic situation right now is when the Link tries to get esbtablishment, it is terminated.

What have you tried to diagnose or workaround this issue?

I tried to use DNS etc. used the configuration used in GSM PPP sample example. Tried various configurations (DNS, increase PPP L2 timeout, ) but no success. I tried also other apps (echo_client and http_get) but they also see issue with socket connection. As offloading mechanism with the modem works fine.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=STM32L475\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

I should see consistent performance whenever I restart the device. but not able to get that yet.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

I want to enable Zephyr TCP/IP stack for the modem and want to see its performance.

**Logs and console output**

[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:26.336,000] [1B][0m<dbg> net_l2_ppp.ppp_fsm_input: (0x20003824): [LCP/0x200007ac] LCP Terminate-Req (5) id 3 payload len 0[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:26.337,000] [1B][0m<dbg> net_l2_ppp.fsm_recv_terminate_req: (0x20003824): [LCP/0x200007ac] Current state OPENED (9)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:26.337,000] [1B][0m<dbg> net_l2_ppp.fsm_recv_terminate_req: (0x20003824): [LCP/0x200007ac] Terminated by peer[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:26.337,000] [1B][0m<dbg> net_l2_ppp.ppp_change_state_debug: (0x20003824): [LCP/0x200007ac] state OPENED (9) => STOPPING (5) (fsm_recv_terminate_req():845)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:26.337,000] [1B][0m<dbg> net_l2_ppp.ppp_fsm_lower_down: (0x20003824): [IPCP/0x20000898] Current state REQUEST_SENT (6)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:26.337,000] [1B][0m<dbg> net_l2_ppp.ppp_change_state_debug: (0x20003824): [IPCP/0x20000898] state REQUEST_SENT (6) => STARTING (1) (ppp_fsm_lower_down():238)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:26.337,000] [1B][0m<dbg> net_l2_ppp.ppp_fsm_close: (0x20003824): [IPCP/0x20000898] Current state STARTING (1)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:26.337,000] [1B][0m<dbg> net_l2_ppp.ppp_change_state_debug: (0x20003824): [IPCP/0x20000898] state STARTING (1) => INITIAL (0) (ppp_fsm_close():210)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:26.337,000] [1B][0m<dbg> net_l2_ppp.ppp_change_phase_debug: (0x20003824): [0x2000077c] phase NETWORK (3) => DEAD (0) (ppp_link_down():118)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:26.337,000] [1B][0m<dbg> net_l2_ppp.validate_phase_transition: (0x20003824): Invalid phase transition: NETWORK (3) => DEAD (0)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:26.337,000] [1B][0m<dbg> net_l2_ppp.ppp_change_phase_debug: (0x20003824): [0x2000077c] phase DEAD (0) => ESTABLISH (1) (lcp_down():235)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:27.460,000] [1B][0m<dbg> net_ppp.ppp_input_byte: Invalid (0x0d) byte, expecting Address[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:27.460,000] [1B][0m<dbg> net_ppp.ppp_change_state: (0x2000328c): [0x20003118] state ADDRESS (1) => START (0)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:36.337,000] [1B][0m<dbg> net_l2_ppp.ppp_fsm_timeout: (0x200061b4): [LCP/0x200007ac] Current state STOPPING (5)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:36.337,000] [1B][0m<dbg> net_l2_ppp.ppp_change_state_debug: (0x200061b4): [LCP/0x200007ac] state STOPPING (5) => STOPPED (3) (ppp_fsm_timeout():105)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:36.337,000] [1B][0m<dbg> net_l2_ppp.ppp_change_phase_debug: (0x200061b4): [0x2000077c] phase ESTABLISH (1) => DEAD (0) (ppp_link_terminated():103)[1B][0m
[1B][1;32muart:~$ [1B][m[1B][8D[1B][J[00:00:36.337,000] [1B][0m<dbg> net_l2_ppp.ppp_link_terminated: (0x200061b4): [0x2000077c] Link terminated[1B][0m

If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, )
 - Toolchain (e.g *** Booting Zephyr OS build zephyr-v2.3.0-2-g906c3367e359  *** SDK, 0.11.2)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.
I tried with zephyr v2.3.9 stable release.

mniestroj, 2020-06-30:
@mtahirbutt Could you try if #26496 helps? I've got similar issues to `net_ppp.ppp_input_byte: Invalid (0x0d) byte, expecting Address` with QEMU, so there is good chance it will help you too.

mtahirbutt, 2020-06-30:
@mniestroj I just tried now, I see ring buffer empty messages, but ppp link is terminated still, the problem is I see DNS server negotiation with something (8.8.8.8) which I saw sometimes with successful PPP connection. I feel like it is more like DNS configuration issue as with offloading I had not to worry but I have to use Zephyr IP stack now correctly. Anyways thanks for the idea.

mniestroj, 2020-06-30:
Looking at beginning of provided log:
```
[00:00:26.336,000] net_l2_ppp.ppp_fsm_input: (0x20003824): [LCP/0x200007ac] LCP Terminate-Req (5) id 3 payload len 0
[00:00:26.337,000] net_l2_ppp.fsm_recv_terminate_req: (0x20003824): [LCP/0x200007ac] Current state OPENED (9)
...
```
it seems that you are receiving termination request from modem. Hard to say why based on current information. Maybe there is something interesting happening in PPP stack before?

Please use \`\`\` before and after logs, so they are nicely rendered in issue description. Remove also ASCII escape sequences, so copy text only.

mtahirbutt, 2020-06-30:
I added another log (a bit detailed), just trying also from different locations may be something odd is happening from the network side or from modem.

mtahirbutt, 2020-07-01:
ok, the issue resolved, I have to set correct CONFIG_MODEM_GSM_APN="wm" which had default "internet". closing it.

closed 2020-07-01 by mtahirbutt