zephyrproject-rtos/zephyr #26513: Improve the disk_access_spi_sdhc for enhancing the performance

rick1082, 2020-06-29:
**Is your enhancement proposal related to a problem? Please describe.**
When trying to connect with a SD card through SPI, disk_access_spi_sdhc.c provides great help.
However, there still something can improve for raising up the throughput performance.
The ```SDHC_SPI_INITIAL_SPEED``` in ```disk_access_spi_sdhc.c``` is 400000, which is very slow.
There's ```SDHC_SPI_SPEED``` which should be used after SDHC initialized, but won't effect after SDHC initialized.

**Describe the solution you'd like**
1. There should be an API for developer to change the ```SDHC_SPI_INITIAL_SPEED```
2. The SDHC_SPI_SPEED should take effect after SDHC initialized, or there should be an API for developer to configure it during runtime. 

**Additional context**
The ```SDHC_SPI_INITIAL_SPEED``` can be found in [here ](https://github.com/zephyrproject-rtos/zephyr/blob/c0d3c8004422b359fd2b6257a2c8e226eff54dd6/subsys/disk/disk_access_spi_sdhc.c#L19
) , and the ```SDHC_SPI_SPEED``` is been assigned in [here](https://github.com/zephyrproject-rtos/zephyr/blob/c0d3c8004422b359fd2b6257a2c8e226eff54dd6/subsys/disk/disk_access_spi_sdhc.c#L650), but does not take effect.

vaussard, 2020-06-29:
Hello. Please have a look to PR #26319 that I opened a few days ago to fix the SPI clock setup in this driver. I guess that it answers your request (`SDHC_SPI_INITIAL_SPEED` is per the SD specification, so I do not see a point in having an API to change it).

It can help if you test the changes, or even do a review since it did not gained much traction up to now.

There is also issue #22906 that it is related to the slow performance of this driver, but someone has yet to work on it.

rick1082, 2020-07-01:
Hello @vaussard , thanks for the great information and nice PR :) I tested the PR and it works great!
However I believe there still needs an API for developer to configure the SPI clock, since 25 MHz SCLK may not suitable for all hardware platform, so I will remain this ticket open :)  
E.g., I'm using nRF5340 now and the SCLK of high speed SPI peripheral can reach to 32MHz, so there still a little space we can push the performance.

vaussard, 2020-07-02:
Hello, thanks for testing.

There is already an API, you can set the maximum frequency in the device tree (this was also added by my PR, since it was ignored before). But it will be limited to 25 MHz, since this is the maximum clock speed in Default Speed mode as per the SD specifications (see Part 1, Physical Layer Specification, see also here: https://www.sdcard.org/developers/overview/bus_speed/). Going higher would require the support of High Speed mode, which is extra work (issue a CMD6 AFAIK). It is also not 100% clear to me if HS mode is supported in SPI mode:
- The specification (section 7.2.14) says "same as SD mode"
- The wikipedia page (https://en.wikipedia.org/wiki/SD_card#Comparison_to_other_flash_memory_formats) lists 50 MHz as being supported only by microSD variant, but not full-size SD (might be wrong of course)

So if I understand your request correctly, you do not need an extra API but support for High Speed mode. Correct me if I am wrong.

rick1082, 2020-07-24:
Hello @vaussard ,
Thanks for the comments :) I took a glimpse on the SD card specification, looks like we need to use CMD6 to switch the card access mode, then we can use higher speed. However I found all SDHC cards on my hands (two class 10, one class 4) are support 32 MHz without using CMD6 for switching mode. 
So it is a dilemma if we are not able to manual configure the speed, but we have to use a slower speed for communication : S
Do you have any suggestions?

danieldegrasse, 2024-02-13:
@rick1082 Seeing as how the SDHC framework now handles raising the SDMMC clock speed (and that speed is configurable) I am going to go ahead and close out this issue. Please add a comment if you'd like it reopened.

closed 2024-02-13 by danieldegrasse